mushorg/conpot #198: Investigate using wireshark to dissect protocols

glaslos, 2015-01-13:
Wireshark has powerful dissectors for many of the protocols of interest to us.
There are a couple of libraries to interface with wireshark: https://github.com/KimiNewt/pyshark and https://github.com/lukaslueg/wirepy

adepasquale, 2015-01-13:
Cool, so there's no even need to wrap the C code directly?

glaslos, 2015-01-13:
If you can feed them the proper data from inside Conpot I assume your assumption is true :)

adepasquale, 2015-07-08:
I don't think calling tshark for every packet is a good idea, anyway this could work:
https://github.com/KimiNewt/pyshark/blob/master/src/pyshark/capture/inmem_capture.py#L69

glaslos, 2015-07-15:
LGTM. You might be able to tie this to our session handler to have a packet consumer per active session.

glaslos, 2026-09-30:
Closing as out of scope

closed 2026-09-30 by glaslos