mushorg/conpot #202: GSoC'15: getting started with Conpot

glaslos, 2015-03-04:
[image]

Conpot is participating under the umbrella of the Honeynet project in the Google Summer of Code 2015. We have three exciting project proposals and are open for other ideas as well.

This issue is targeted at potential GSoC students to help them getting started and write a **strong proposal**. If there are any questions, please don't hesitate to comment below :smiley:

Picking the right students for a Google Summer of Code is hard - we really want to encourage people to get involved into open source, but how do we determine who's best suited for a particular project? Here comes your part: Show us that you're super enthusiastic, autonomous and good at coding! :tada: 
Most importantly, your proposal should demonstrate two things: You have the ability to execute and you truly believe that Conpot is an interesting project to work on.

### Demonstrate that you are capable of fulfilling the project!

While many of you probably have a good theoretical background from university, you should show us that you can code as well:

Do you have a cool/interesting side project or class project? Show it to us! Even better if it's open source and we can read the code!

* Have you ever contributed to an open source project? Show us what you did!
* No previous projects you could show? Add a feature and/or fix a bug in Conpot!

### Demonstrate that you really want to contribute to Conpot!

> The brick walls are there for a reason. The brick walls are not there to keep us out; the brick walls are there to give us a chance to show how badly we want something. The brick walls are there to stop the people who don't want it badly enough. They are there to stop the other people! [[1]](https://www.youtube.com/watch?v=ji5_MqicxSo)

This year, there are 137 accepted GSoC organizations, each of them with multiple projects. Now, why do you want to work on Conpot specifically? We do not just want to hear a chorus of praise here ("Conpot is a great tool"). Instead, play around with it and tell us about your experiences:

* Modify the default template or create a new one.
* Add a proxy module to real hardware.
* Connect Conpot to your Arduino via Modbus.
* Pick one of the listed issues and fix it.
* Run a vulnerability scanner against an running Conpot instance.

Again, a good way to show that you're a good candidate is to fix a bug, propose or implement a feature, add something that was unclear to you to the docs, etc.

### Get in touch!

We hang out on IRC (#gsoc-honeynet and #conpot on Freenode), then there is the mailing list and of course you can just comment here. Try to ask us good questions! :smiley:

### Good starting points for a contribution

Contributing to Conpot directly is an optional part of your proposal. If you can convince us by other means that you'd be a strong student (e.g. strong past experience with other open source projects), that's good, too.

* Fix bugs or add features. Make sure to filter the issue list by "uncomplicated", which includes suitable tasks.
* Add some docstrings to the code! Read a bit through Conpots code base, take a somewhat complex undocumented class/function, find out what it does and add a quick doc string explaining what it is.
* Improve the docs. If you're proficient in English, feel free to make our docs better! If you're not a native speaker, try to focus on the other tasks though. :smiley:

katkad, 2015-03-04:
:sparkles:

creolis, 2015-03-04:
Big :)

adityasriv22, 2017-11-14:
Looking forward to contribute :)

xandfury, 2018-08-02:
I took the liberty of [migrating this issue to wiki](https://github.com/mushorg/conpot/wiki/GSoC:-Getting-started-with-Conpot!). I hope that's okay.

closed 2018-08-02 by xandfury