mushorg/conpot #209: MODBUS implementation not working correctly

adarshdinesh, 2015-03-17:
Conpot uses python modbus implementation 'modbus-tk' to imitate the modbus TCP server. There is a tool modscan (https://code.google.com/p/modscan/) which was published in DefCon used for scanning MODBUS servers. This tool failed to detect our modbus server. Please check the output 

modscan output : http://pastebin.com/kjPwqbV4
conpot log: http://pastebin.com/hMq4eii1

adarshdinesh, 2015-03-18:
But at the same time, plcscan detects it if we use it with --brute-uid. That is, the slave identifier(1 byte value in TCP MODBUS frame) is bruteforced to find the devices. 
[image]

adarshdinesh, 2015-03-18:
I examined the modscan.py code and understood how they managed to find the modbus. They set a Function code in each request and check if the response also has the same Function code (7th Byte). The tool is not detecting our server because out server isn't sending the Function code back.
https://github.com/adarshdinesh/modscan/blob/master/modscan.py#L156
This can be a trap .. What if the actual modbus TCP servers send the same Function code back and our modbus-tk implementation does it in wrong way ? We have to check this

adarshdinesh, 2015-03-18:
./modscan.py -d --function=0 127.0.0.1
returned  [0, 0, 0, 0, 0, 3, 246, 128, 4]
7th byte is Function code 128

./modscan.py -d --function=2 127.0.0.1
[0, 0, 0, 0, 0, 3, 246, 130, 4]
7th byte is 130 

I have no idea why our modbus implementation add a 128 and send the Function code back. Only because of that reason modscan says its not a real modbus implementation

adarshdinesh, 2015-03-18:
http://www.simplymodbus.ca/exceptions.htm
It's an Error Response : "The first sign of an exception response is that the function code is shown in the echo with its highest bit set". So the modscan scanner tool is checking if the response is an exception or not and if its an exception then it will say that its not a modbus server.
And our conpot log also says it was an exception.  

2015-03-19 04:11:18,144 Exception caught: Modbus Error: Exception code = 3. (A proper response will be sent to the peer)
2015-03-19 04:11:18,145 Client disconnected. (98291a02-8168-48a7-a3c0-372b0d88eab7)
2015-03-19 04:11:18,145 New connection from 127.0.0.1:58147. (98291a02-8168-48a7-a3c0-372b0d88eab7)
2015-03-19 04:11:18,145 Exception caught: Modbus Error: Exception code = 3. (A proper response will be sent to the peer)

adarshdinesh, 2015-03-18:
We will make modscan believe that it's a real modbus TCP server by resetting the MSB bit of function code. May be we can do it from the proxy script.

glaslos, 2015-03-19:
Have you checked why we are sending a error code?
You can also have a look at the modbus version which we are using: https://github.com/glastopf/modbus-tk

adarshdinesh, 2015-03-19:
https://github.com/glastopf/modbus-tk/blob/master/modbus_tk/modbus.py#L357
This is the line where its happening I think.  Checking

infracritical, 2015-03-31:
Working with the ICS/SCADA community, the 'modscan' tool is riddled with bugs.  I know the author very well, and we have had numerous discussions about his tool.

For those who want to perform a MODBUS scan, I would suggest using Digital Bond's NMAP NSE script as part of their Project Redpoint.

The NSE script may be found here:

https://github.com/digitalbond/Redpoint/blob/master/modicon-info.nse

Also, for those of you who want to know more about ICS/SCADA security, we own and sponsor a publicly and freely available mailing list to the entire world.  It's called "SCADASEC", and is the largest and most active of any ICS/SCADA security discussion forums out there.  One word of caution though...the IC groups from over 32 countries, including military, *DO* watch our list...so please be careful as to what you type... ;)

infracritical, 2015-03-31:
DOH!  Forgot the URL to the mailing list...

http://www.scadasec.com

infracritical, 2015-03-31:
One last thing...

Although the NMAP NSE script was written primarily for the Schneider-Electric/Telemechanique/MODICON devices, it *should* work 'generically' on *any* MODBUS device.

Food for thought... ;)

glaslos, 2015-04-01:
Like here: https://github.com/glastopf/conpot/issues/169 ? ;)

Vingaard, 2015-04-03:
Please see update on https://github.com/glastopf/conpot/issues/169

creolis, 2017-06-28:
This should be fixed now, I'll validate ..

windflyer, 2017-08-27:
@creolis I downloaded modscan following the googlecode link. It successfully detected our modbus server running with the default template. This issue has been addressed.

glaslos, 2017-08-27:
@windflyer thanks for checking!

closed 2017-08-27 by glaslos