mushorg/conpot #211: Remove Interpolation overhead from logging scripts

adarshdinesh, 2015-03-21:
There are a lot of unwanted interpolations done in the conpot logging scripts. This should be removed.

sandbox@sandbox:~/conpot$ grep -r ".format(" | grep "logg" > lines

http://pastebin.com/pxb3gff0

adarshdinesh, 2015-03-21:
It looks like somebody should manually do it  for some of the lines. I just wrote a small script to automate and rewrite the line with right script, but the script failed in some cases. 
Example : build/lib.linux-x86_64-2.7/conpot/protocols/snmp/snmp_server.py:184:        logger.info('SNMP server started on: {0}'.format((host, self.get_port())))
Where the function is directly called from the logger and our parser script will fail in such places.
Script Output : http://pastebin.com/GBiiarXn

glaslos, 2015-03-22:
Just create a pull request for the changes ;)

adarshdinesh, 2015-03-24:
https://github.com/adarshdinesh/conpot/commit/f695ac81d94b083eff7bb23c02a1ab3a9a38146f
Please check it.

adarshdinesh, 2015-03-24:
https://github.com/adarshdinesh/fixinterpolation

12 need manual work :(
-log-
File conpot/protocols/s7comm/s7_server.py line number [111]
File conpot/protocols/kamstrup/meter_protocol/messages.py line number [65]
File conpot/protocols/modbus/modbus_server.py line number [49]
File conpot/protocols/kamstrup/meter_protocol/command_responder.py line number [49]
File conpot/protocols/snmp/snmp_server.py line number [137, 164]
File conpot/emulators/proxy.py line number [65]
File conpot/protocols/snmp/conpot_cmdrsp.py line number [78, 87]
File conpot/protocols/http/command_responder.py line number [1046]
File conpot/protocols/kamstrup/meter_protocol/decoder_382.py line number [78, 116]

adarshdinesh, 2015-03-24:
71 problems fixed :) Repository health increased 90%. Please merge changes.

glaslos, 2015-03-24:
Waiting for a separate pull request. Great work!

adarshdinesh, 2015-03-24:
63 problems Fixed. The file s7_server.py is untouched as we are working on a bug in that file on the same lines. Pull Req : https://github.com/glastopf/conpot/pull/222

conpot/protocols/s7comm/s7_server.py:58
conpot/protocols/s7comm/s7_server.py:67
conpot/protocols/s7comm/s7_server.py:111
conpot/protocols/s7comm/s7_server.py:172
conpot/protocols/s7comm/s7_server.py:177
conpot/protocols/s7comm/s7_server.py:182

glaslos, 2015-03-25:
Fixed in #222

closed 2015-03-25 by glaslos