mushorg/conpot #212: How do I customize a new template

infracritical, 2015-03-23:
Am trying to create another template for the honeypot.  Have duplicated the /opt/conpot/templates/default directory to /opt/conpot/templates/new_template.

Upon attempting to execute: sudo conpot --template new_template &

I get the error message: "Template not found"

I've tried to utilize the existing 'default' template.xml file, and customized the entries within it to get rid of the "Technodome" example into something more meaningful.  However, when attempting with initiating the abovestated CLI entry, upon checking the web entry, it STILL shows "Technodome".

What am I doing wrong???

I want to create multiple templates, all using varied Siemens SIMATIC S7 models, and define addition datapoint entries for testing purposes to respond back to MODBUS queries.

Thoughts anyone???

-rad

P.S.  I've looked extensively for customizing a *new* template, and only found the original URL; again, when attempting in creating a new template, it still shows "Technodome"

infracritical, 2015-03-24:
Found the bloody files for 'production'.  It's located under here:

/usr/local/lib/python2.7/dist-packages/Conpot-0.4.0-py2.7.egg/conpot/templates

...and NOT here:

/opt/conpot/conpot/templates

In the immortal words of Homer Simpson....DOH!!!!  Closing...

katkad, 2015-03-25:
you can also add template to /opt/conpot/conpot/templates but then you have to install conpot once more (python setup.py install). templates will copy as well to /usr/local/lib/python2.7/dist-packages/Conpot-0.4.0-py2.7.egg/conpot/templates

infracritical, 2015-03-25:
Kind of figured that much out already (see previous comment from me)...thanks.

Closing request...

closed 2015-03-25 by infracritical