mushorg/conpot #213: Alternative manufacturers?

infracritical, 2015-03-23:
Anyone working on other manufacturers?  Just curious...everyone seems to want to use ONLY Siemens SIMATIC devices.  Is there any other effort -- outside of maybe Digital Bond -- working on manufacturers, say Rockwell or Schneider Electric?

glaslos, 2015-03-24:
Digital Bond is working on Conpot profiles?
Creating new profiles is a rather complex task. Most of the time we don't have access to real hardware which is required to model the template.
We also struggle finding a solution to provide HMI templates without violating vendor copyrights.

infracritical, 2015-03-24:
Digital Bond uses a variant of 'honeywall' and/or 'honeyd'; however, they may use 'conpot' -- need to ask Digital Bond re: question.

infracritical, 2015-03-24:
Also, have access to several devices, as well as other researchers that we work closely with.  Am thinking of a few vendors, starting with Rockwell Automation/Allen-Bradley and Schneider-Electric/Telemecanique/MODICON.  Have good relationships with both vendors; will ask.

glaslos, 2015-03-24:
Great! Let me know if you need additional help. We are also reachable on #conpot@freenode. Closing this for now.

closed 2015-03-24 by glaslos