mushorg/conpot #215: Unnoticed Bug in s7_server.py

adarshdinesh, 2015-03-24:
This bug was there in the code for a long time. This came out when I remove the unwanted interpolations from logging scripts : https://github.com/glastopf/conpot/blob/master/conpot/protocols/s7comm/s7_server.py#L116

adarshdinesh, 2015-03-24:
Nothing much. An unnoticed bug and an extra line.

glaslos, 2015-03-24:
Pull request?

adarshdinesh, 2015-03-24:
https://github.com/glastopf/conpot/pull/217

adarshdinesh, 2015-03-24:
https://github.com/glastopf/conpot/pull/220

adarshdinesh, 2015-03-24:
Pull Req : https://github.com/glastopf/conpot/pull/223 
This issue is fixed :) Please merge it so that we can do the remainging Interpolation removal from the s7_comm.py file.

adarshdinesh, 2015-03-25:
@glaslos : Please check this : https://github.com/glastopf/conpot/pull/224

glaslos, 2015-03-25:
Fixed in #224

closed 2015-03-25 by glaslos