mushorg/conpot #225: Error while connecting to proxy server

harisphnx, 2015-03-26:


harisphnx, 2015-03-26:
I installed the conpot server, and ran it with the proxy template. But when trying to connect to the Kamstrup Channel A proxy server. The connection gets refused, after that a number of times i get this error message..

Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/gevent-1.0.1-py2.7-linux-x86_64.egg/gevent/greenlet.py", line 327, in run
    result = self._run(*self.args, **self.kwargs)
  File "/usr/local/lib/python2.7/dist-packages/gevent-1.0.1-py2.7-linux-x86_64.egg/gevent/server.py", line 102, in wrap_socket_and_handle
    ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
  File "/usr/local/lib/python2.7/dist-packages/gevent-1.0.1-py2.7-linux-x86_64.egg/gevent/ssl.py", line 383, in wrap_socket
    ciphers=ciphers)
  File "/usr/local/lib/python2.7/dist-packages/gevent-1.0.1-py2.7-linux-x86_64.egg/gevent/ssl.py", line 87, in __init__
    cert_reqs, ssl_version, ca_certs)
SSLError: [Errno 336265218] _ssl.c:355: error:140B0002:SSL routines:SSL_CTX_use_PrivateKey_file:system lib
<Greenlet at 0x7fa29bb58230: <bound method StreamServer.wrap_socket_and_handle of <StreamServer at 0x7fa29bbb6b50 fileno=9 address=0.0.0.0:1234 handle=<bound method Proxy.handle of <conpot.emulators.proxy.Proxy object at 0x7fa29bbb6b90>>>>(<socket at 0x7fa29bb67890 fileno=15 sock=10.100.55, ('10.100.53.104', 2422))> failed with SSLError

I read a bit on the internet, it seems that a number of ssl related issues are mapped to this common exception, so it might be difficult to pin point the reason.
At one place, i read the server might not have permission to read the ssl key file, but i ran the server in sudo mode..

glaslos, 2015-11-08:
We are dropping permissions after the start, but the server should read the key into memory before we drop them. So not quite sure whats going on. Closing this until we can reproduce.

closed 2015-11-08 by glaslos