mushorg/conpot #226: Website Creation with login feature in combination with conpot

mav3rick1990, 2015-03-27:
Hi,

I am doing a semester project in which I am trying to implement conpot and also create a webserver hosting some webpages for the attacker to believe they have hit something. This website will host certain miscellaneous data and simulated PLC from conpot. 

I understand that conpot runs its own apache server and I am able to upload web docs to htdocs folder as well. This displays my initial login page. I am now trying to implement an authentication method to check for username and passwords. Thought about installing LAMP since it includes apache sql and php support. However, I know that i cannot use apache since im using conpot. Is there a way to work around this? 

Any guidance would be of much help. 

Cheers!

creolis, 2015-03-27:
You can proxy your requests through conpot.
Try this: Run your apache on port 8080, bind it to localhost. Then configure conpot to proxy all requests that are not fulfilled by conpot itself, by applying the PROXY config directive to the STATUS-404 configuration directive. Point the proxy directive to localhost:8080. 

Boom -> you're there :)

glaslos, 2015-04-13:
Closing for now, assuming this is answered.

closed 2015-04-13 by glaslos