mushorg/conpot #230: CentOS Support?

infracritical, 2015-03-30:
Want to know if anything OTHER THAN Ubuntu is supported.  Will this work with RHEL or CentOS?

glaslos, 2015-03-30:
I think it works even with Windows ;-)

infracritical, 2015-03-30:
Nice.  OK, then...time to try it out...will advise based on my results.  Thanks!

infracritical, 2015-03-31:
Nope...have to compile most of this from source...lovely, just lovely...

glaslos, 2015-04-14:
Did you manage to run Conpot on CentOS?

infracritical, 2015-04-14:
No.  CentOS uses an older version of Python to perform updates for the operating system, primary for it's *yum* and *rpm* update processes.  if I try and update Python to 2.7.x, it boloxes the update process, breaking it.  This has been an area of consternation with CentOS and Fedora developers.

I simply gave up, but am thinking of re-doing the effort again.

If I made a stable CentOS environment, say running Version 6.6 that is patched, would it help to use that as a *sandbox*?  I have 40 static IP addresses, and can dedicate a server specifically for this.

Do know...*all* of my develop is using 32-bit architecture, as much of it is donated or scrapped equipment obtained for research purposes.  Additionally, 32-bit systems are *still* used for ICS/SCADA environments for many asset owners of varied infrastructure sectors.

For example, the Energy Sector is *very* reluctant in updating or utilizing newer architecturs, as this would require a *full* ECR be performed as required by either NRC or NERC regulations.  The process of re-certifying any given *production* system, no matter how old that it may, would take months, sometimes upwards of a year, to satisfy regulation requirements.

This is an area of frustration for many security researchers and developers here within the U.S.

infracritical, 2015-04-14:
I've re-opened this issue, as I think it important enough to work out a solution for Red Hat (RHEL, CentOS, Fedora) environments, as these architectures are used quite heaviliy within production environments here within the U.S.

Spindel, 2015-04-14:
I'd suggest using scl and https://www.softwarecollections.org/en/ to run
python 2.7 on CentOS 6. A slightly modified Init script to set up the
environment is needed though

adepasquale, 2015-04-14:
> CentOS uses an older version of Python to perform updates for the operating system, primary for it's yum and rpm update processes. if I try and update Python to 2.7.x, it boloxes the update process, breaking it.

Sorry for my lack of knowledge about CentOS, but is it a viable option to use a virtualenv?

infracritical, 2015-04-14:
Nice.  However, I noticed *one* glitch: the version of Python is Version 3.3.  Isn't 'conpot' tested *only* for Version 2.7.x?  If memory serves me right, Version 3.3 is the current development version and may not be stable enough for this effort.  I would rather use a more stable release to base any development on a Red Hat architecture (RHEL, CentOS, Fedora).

glaslos, 2015-04-14:
CentOS is definitely a platform we want to support.
Conpot is only tested for Python 2.7
Python 3.x is not an option for now as many 3rd party libraries we are using are still on 2.x
Compatibility with Python 2.6 might be an option.
Installing Python 2.7 side-by-side with Python 2.6 on CentOS might be an option too: https://github.com/h2oai/h2o/wiki/Installing-python-2.7-on-centos-6.3.-Follow-this-sequence-exactly-for-centos-machine-only (expect to go through some pain...)

infracritical, 2015-04-14:
I experienced this several times in the past when attempting to implement several Python solutions that used Version 2.7.  I think it best if I create a 'sandbox' environment for this effort.

Spindel, 2015-04-14:
https://www.softwarecollections.org/en/scls/rhscl/python27/ gives a good side by side.

infracritical, 2015-04-14:
Fair enough.  Good place to start...thank you.

creolis, 2015-04-14:
>> [...] Version 3.3 is the current development version [...]

I would like to encourage you to think of Python3 as "a new Python" rather than an "updated version" since it is not really compatible and brings a lot of substantial changes.

I'm sure some people will not share this opinion, but I think Python 2.x vs. Python 3 can be compared to C vs. C++ or Apache 1.3 vs. Apache 2 in terms of "not just a minor upgrade".

glaslos, 2015-04-14:
@creolis I don't share your opinion :) For me 3.x is the better Python!

closed 2015-06-01 by infracritical

creolis, 2015-06-01:
@glaslos I didn't state that it is not a better version ... I just said it's more than a simple successor :)