mushorg/conpot #231: A simple question about kamstrup

KostasChar, 2015-04-09:
Hi i am doing a project about honeypots in my college and i installed conpot.I was interested in smart grid security and i installed kamstrup as well.I also used nmap to see if it was working properly.And the result was port 1025/tcp open and service NFS-or-IIS.
My question is how a potential attacker can understand that it is a smart meter and not something ordinary.Even the -O  flag didn't show anything..
Moreover are smart meters going to be exposed to the public internet,cause i thought they were controlled by the department of energy of each country and that there is a separate network infrastructure.In my country we haven't implemented these technologies and the idea of smart meters with public ip is somewhat confusing to me.I mean is there a practical consideration about this project or is it research oriented?
Thanks in advance!!!

czardoz, 2015-04-09:
1. Potential attackers will also try to send some packets to the open port, which will respond as a smart meter, allowing them to "discover" it.

2. Some of them might have public IPs.

KostasChar, 2015-04-09:
Thanks mate!keep up the good work!

czardoz, 2015-04-10:
A few links to help:

* https://www.youtube.com/watch?v=8QwpcM0L3x0
* http://conpot.org/
* http://www.honeynet.org/node/1047

glaslos, 2015-04-13:
@KostasChar we are currently working on solving the nmap -O issue: https://github.com/glastopf/oschameleon

Closing the issue for now, assuming all questions answered.

closed 2015-04-13 by glaslos