mushorg/conpot #232: DNP3_Proxy - Install Doc. Update

AGIsmail, 2015-04-10:
Hi,

Thought I'd add a couple of additional steps to < conpot/docs/source/components/protocols/dnp3.rst >.
Might be helpful to some.

=-Qu
1) Fetch DNP3 2.0.x branch from https://github.com/automatak/dnp3 i.e. git clone https://github.com/automatak/dnp3.git
2) Install the prerequisites: http://dnp3.github.io/doc/2.0.x/building/prerequisites.html
3) Get ASIO from here: http://think-async.com/ and install it (./configure/make/make install)
4) Before building dnp3 change the following file: dnp3/cpp/examples/master/DemoMain.cpp

Change the port on line 54 from 20000 to 20001, so that it becomes:
	auto pChannel = manager.AddTCPClient("tcpclient", FILTERS, TimeDuration::Seconds(2), TimeDuration::Seconds(5), "127.0.0.1", "0.0.0.0", 20001);

5) Follow the build instructions: http://dnp3.github.io/doc/2.0.x/building/building-linux.html
6) Run conpot with the proxy i.e. conpot -t dnp3_proxy.xml
7) Run ``outstationdemo`` and ``masterdemo``. 

Note: If you build before the modification (step 3) proceed with the following:

1) Run: find /your/dir -type f -exec touch {} +
2) Run: make clean
3) Continue with steps 4 to 6 above.

If you get a warning concerning a clock skew while running these additional steps, you may ignore it since you're running a clean build.
=-Uq

Changes:
1) Removed http://dnp3.github.io/doc/2.0.x/building/prerequisites-linux.html <- this link no longer exists
2) Added steps 3 and 5 and the clean build notes.

Any feedback is more than welcome.

glaslos, 2015-04-10:
You should send this in as a pull request! Looks good!

AGIsmail, 2015-04-10:
Glad you like it :)
Pull request issued: #233.

infracritical, 2015-04-17:
Um....question...

Is utilizing this as a *proxy* the ONLY method, or can this configured with DNP3 implemented the same way that S7COMM and MODBUS are configured?

glaslos, 2015-04-18:
@infracritical the only way would be if we had a Python wrapper for DNP3 (SWIG, ctypes, pyrex, SIP or if you wanna go badass Boost...).

glaslos, 2015-11-08:
Closed with #233

closed 2015-11-08 by glaslos