mushorg/conpot #234: MAC masquerading not working

infracritical, 2015-04-13:
I am attempting to masquerade the MAC address to the primary NIC where the device registers as something else.

If I perform the following commands initially:

ifdown eth0
ifconfig eth0 hw ether 00:11:22:33:44:55 (example here, not real MAC)
ifup eth0

It doesn't allow the 'conpot' instance to come up.  OR...if it does, the web server doesn't appear to come up.

if I perform the same commands AFTER the 'conpot' instance is started, while attempting to interrgate the 'conpot' device, it becomes unresponsive for ALL services, not just the web server.

What am I doing wrong here?

===========

P.S.  I've tried this under a virtualized environment, and get similar results, and an NMAP scan still reveals the virtualized server's virtual NIC.

glaslos, 2015-04-14:
Have you tried this: https://github.com/glastopf/conpot/blob/master/conpot/conpot.cfg#L49 ?

infracritical, 2015-04-14:
That depends...

if it is on a 32-bit server, then it's simply the HTTP service, and the remaining services (MODBUS, SNMP, S7COMM) continuing running.

If it is on a 64-bit server, then the entire environment is 'flakey', and I get mixed results, sometimes only the HTTP services, other times, ALL services -- become unresponsive.

Thoughts?

glaslos, 2015-04-14:
This is all in a VM?

infracritical, 2015-04-14:
Yes and no.

The test is being conducted on *both* a virtualized environment AS WELL as a dedicated server.  see previous comments about results.  I get similar 'flakey' results on both.

glaslos, 2015-04-14:
Ah, sorry, I missed your post scriptum.
I'll try to reproduce your issues tonight. Changing the MAC address is a core feature we want to have for Conpot, so I consider this to be rather critical.

infracritical, 2015-04-14:
Thanks.  Did you want me to log the process?  I could easily provide raw output for dissemination here.

glaslos, 2015-04-14:
Anything you are able to share might help :)

infracritical, 2015-04-14:
Absolutely!  I'll see what I can do this evening, and post my results here.  ;)

infracritical, 2015-05-07:
Sorry for the long delay...been sucked into another project.  I will do the logging again, and post here when I have time -- hopefully -- this weekend...

glaslos, 2015-05-08:
Hm, I can reproduce this. I'm investigating...

glaslos, 2015-05-08:
Are you by any chance on dhcp? If so, you need a new lease after changing the MAC address.

glaslos, 2015-05-08:
Try the latest version from git. You have to change the config and set `change_mac_addr` to True.

infracritical, 2015-05-11:
I am currently using v0.4.0, and get similar results.

glaslos, 2015-05-11:
0.4.0 is behind the master branch. We are going to release a new version after merging the code from @sookyp 
If you want to try it with 0.4.0: Are you using DHCP? Have you tried to get a new lease?

infracritical, 2015-05-28:
Sorry for the long wait, no, not using DHCP; IPs are statically assigned.

infracritical, 2015-06-01:
Found a workaorund that seems to work.  However, the masquerading via the "conpot.cfg" file is still unable to work effectively, and produces irregular results.  More to come...

closed 2015-06-01 by infracritical