mushorg/conpot #235: Installation

shreyas1989, 2015-04-21:
Hi I get the following error, 

Collecting crc16 (from conpot)
  Using cached crc16-0.1.1.tar.gz
Collecting enum (from conpot)
  Using cached enum-0.4.4.tar.gz
Collecting hpfeeds (from conpot)
  Could not find a version that satisfies the requirement hpfeeds (from conpot) (from versions: )
  Some externally hosted files were ignored as access to them may be unreliable (use --allow-external hpfeeds to allow).
  No matching distribution found for hpfeeds (from conpot)

glaslos, 2015-04-21:
Try this:

```
git clone https://github.com/rep/hpfeeds.git
cd hpfeeds
sudo python setup.py install
```

closed 2015-04-21 by glaslos

shreyas1989, 2015-04-21:
ok thanks.. will try this :+1: