mushorg/conpot #237: utf problem in log worker

johnnykv, 2015-04-25:
While setting up some ConPot honeypots, I experienced a crash in the LogWorker class. This can be reproduced by telnetting to the Kamstrup management implementation.

```
2015-04-25 21:26:05,297 Kamstrup management traffic from 87.104.18.136: {'request': 'h\r\n', 'response': '==============================================================================\r\nService Menu\r\n==============================================================================\r\nH:   Help [cmd].\r\nQ:   Close connection.\r\n!AC: Access control.\r\n!AS: Alarm Server.\r\n!GC: Get Config.\r\n!GV: Software version.\r\n!SA: Set KAP Server IP and port (*1).\r\n!SB: Set 2nd KAP Server IP and port.\r\n!SC: Set Config (*1).\r\n!SD: Set device name (*1).\r\n!SH: Set KAP Server lookup (DNS or DHCP)\r\n!SI: Set IP (enter either valid IP or 0 to force DHCP)(*1).\r\n!SK: Set KAP watchdog timeout(WDT).\r\n!SN: Set IP for DNS Name servers to use.\r\n!SP: Set IP Ports\r\n!SS: Set Serial Settings.\r\n!RC: Request connect\r\n!RR: Request restart (*1).\r\n!WM: Wink module.\r\n==============================================================================\r\n(*1) Forces system restart\r\n==============================================================================\r\nKamstrup (R)\r\n'} (d1d4f3f5-52b9-44a4-9f6b-992f71aaacd0)
2015-04-25 21:26:11,401 Kamstrup management traffic from 87.104.18.136: {'request': '\xff\xf4\xff\xfd\x06', 'response': "\r\n? Command not found.\r\nSend 'H' for help.\r\n"} (d1d4f3f5-52b9-44a4-9f6b-992f71aaacd0)
2015-04-25 21:26:11,414 Stopping because <Greenlet at 0x7f602da1fc30: <bound method LogWorker.start of <conpot.core.loggers.log_worker.LogWorker object at 0x7f603344bad0>>> died: 'utf8' codec can't decode byte 0xff in position 0: invalid start byte
```

adepasquale, 2015-04-30:
HPFeeds is a simple binary protocol. To my knowledge, data would be packed this way:
https://github.com/rep/hpfeeds/blob/249b2f7f44e6e017f5139e3f074fa0710e0a830c/lib/hpfeeds.py#L25-30

So, how about simply adding `ensure_ascii=False` to `json.dumps()` here?
https://github.com/glastopf/conpot/blob/master/conpot/core/loggers/log_worker.py#L129

katkad, 2015-07-30:
I got different error in both with and without proposed fix

```
2015-07-30 11:53:44,821 Kamstrup management traffic from 172.20.7.135: {'request': '\xe6\xb1\x89\xe8\xaf\xad / \xe6\xbc\xa2\xe8\xaa\
x9e;\r\n', 'response': "\r\n? Command not found.\r\nSend 'H' for help.\r\n"} (60147129-dd25-4dd3-9ead-506c921d4b5c)
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/gevent-1.0.1-py2.7-linux-i686.egg/gevent/greenlet.py", line 327, in run
    result = self._run(*self.args, **self.kwargs)
  File "/usr/local/lib/python2.7/dist-packages/Conpot-0.4.0-py2.7.egg/conpot/core/loggers/log_worker.py", line 132, in start
    self.sqlite_logger.log(event)
  File "/usr/local/lib/python2.7/dist-packages/Conpot-0.4.0-py2.7.egg/conpot/core/loggers/sqlite_log.py", line 63, in log
    event["data"].get('request'), event["data"].get('response'))
ProgrammingError: You must not use 8-bit bytestrings unless you use a text_factory that can interpret 8-bit bytestrings (like text_f
actory = str). It is highly recommended that you instead just switch your application to Unicode strings.
<Greenlet at 0xb4c26edcL: <bound method LogWorker.start of <conpot.core.loggers.log_worker.LogWorker object at 0xb4bd2eac>>> failed 
with ProgrammingError

ERROR:root:Stopping because <Greenlet at 0xb4c26edcL: <bound method LogWorker.start of <conpot.core.loggers.log_worker.LogWorker obj
ect at 0xb4bd2eac>>> died: You must not use 8-bit bytestrings unless you use a text_factory that can interpret 8-bit bytestrings (li
ke text_factory = str). It is highly recommended that you instead just switch your application to Unicode strings.
2015-07-30 11:53:45,870 Stopping because <Greenlet at 0xb4c26edcL: <bound method LogWorker.start of <conpot.core.loggers.log_worker.
LogWorker object at 0xb4bd2eac>>> died: You must not use 8-bit bytestrings unless you use a text_factory that can interpret 8-bit by
testrings (like text_factory = str). It is highly recommended that you instead just switch your application to Unicode strings.
```

connect to kamstrup mngmt interface wth telnet, and type one of these:
České dráhy
汉语 / 漢語;

johnnykv, 2015-09-08:
I can't reproduce this problem anymore. @katkad @adepasquale do you of  you have a bulletproof way to reproduce these two crashes?

johnnykv, 2015-09-12:
Please reopen if this is still an issue.

closed 2015-09-12 by johnnykv

katkad, 2015-10-13:
nice, works fine now