mushorg/conpot #238: HTTP webserver not working

mav3rick1990, 2015-04-27:
Hello!

I am trying to implement conpot on one of my machines. However, I am unable to see the default webpage "Technodrome" using a web browser unless I get an external IP address for the machine. I do not wish for this machine to be facing the internet therefore I had disabled fetch external ip option in conpot.cfg.

I am only able to see the open http port when I connect back to the internet allowing conpot to get a public IP address. 

Hoping to find an explanation as to why this is necessary. It will be a great help for my project.

Thank you.

creolis, 2015-04-27:
Would you please start conpot, run

```netstat -an | grep LISTEN```

in another terminal and let us see the output?

mav3rick1990, 2015-04-27:
As requested.

[image]
[image]

creolis, 2015-04-27:
hmm ... that should not happen and is definitely worth investigating.

To your question:

> I do not wish for this machine to be facing the internet therefore I had disabled fetch external ip option in conpot.cfg.

I'm afraid you misunderstood this option :) It will not magically get you an external IP address and expose your setup to the internet - all this option actually does is that it DETECTS your external IP address in order to get a number of services (logging, etc.) configured the right way. Maybe we should rename this parameter to "detect_external_ip" though as this seems misleading :)

So you don't have to worry about that - you can enable the detection mechanism again and use conpot happily ever after, but I will leave this issue open as we should investigate why the web server does not come up without IP detection and shows no failure record.

glaslos, 2015-04-27:
Hm, this is odd...

creolis, 2015-04-27:
@glaslos: yeah, .. I didn't even know of some kind of link between fetching the WAN IP and spawning our services (except for hpfriends and stuff that may rely on WAN IP .. and even they should mitigate that .. )

mav3rick1990, 2015-04-28:
"So you don't have to worry about that - you can enable the detection mechanism again and use conpot happily ever after,"

Hi there! just want to ensure I have explained the situation properly for rectification. Even if I enable fetch external ip and run conpot, it won't work until my computer is actually connected to the internet. Below are the screenshots for :

when i am not connected to the internet but external ip is enabled
[image]
when I connect to the internet and run conpot.
[image]

glaslos, 2015-05-08:
I can't find any connection between the ext_ip code and the HTTP protocol code. I assume the issue is somewhere else... Maybe the lack of an internet connection?

wojtossfm, 2015-10-26:
@mav3rick1990 Not that it's a legitimate solution to the problem but having a dummy localhost HTTP server that will respond even with 127.0.0.1 to a single request and shutdown seems to help circumvent the problem. Hope this helps while looking for a viable solution.

glaslos, 2015-10-26:
@wojtossfm is http the only protocol that's not working?

wojtossfm, 2015-10-26:
@glaslos Seems like it. With fetch_public_ip set to off I can telnet to the s7comm and modbus servers and using pysnmp examples I can connect to the SNMP server. Only HTTP doesn't work (doesn't even showup in netstat).

glaslos, 2015-10-26:
@wojtossfm does your Conpot has an internet connection?

wojtossfm, 2015-10-26:
It does have an internet connection, but I didn't want it to initiate connections to the interet and that is why I was looking for a way to disable fetching the public ip.

glaslos, 2015-10-26:
Fair point. @creolis do you have some time to look into this? I am pretty busy the next two weeks.

creolis, 2015-10-26:
yeah, I'll have a look at it ...

wojtossfm, 2015-10-26:
Seems like this might be some form of race in initialization. Adding a sleep when fetch is disabled also seems to solve the issue...
```
    public_ip = None
    if config.getboolean('fetch_public_ip', 'enabled'):
        public_ip = ext_ip.get_ext_ip(config)
    else:
        import time
        time.sleep(1)
```

glaslos, 2015-10-26:
@wojtossfm I don't see an issue with removing the `else` and outdent the sleep (put the import to the top of the file). Are you comfortable with creating a PR for this?

wojtossfm, 2015-10-26:
@glaslos No problem. I can make a PR with the changes you mentioned. The import was placed there only to make the sample code more compact.

wojtossfm, 2015-10-26:
@glaslos I just realized that it seems that this was fixed somewhere along the way without anyone noticing directly. I was working with a 0.4 release version and just now I tried with a fresh 0.4 release version (I noticed they were different) and the bug isn't there. I'll check up on tomorrow to make sure  as I have to call it a day already.

Sorry for possibly taking up your time. Didn't realize a silent fix could have gone into release but should have checked prior to bringing up the topic. At least possibly if this is confirmed then you'll be able to close the issue.

glaslos, 2015-10-26:
@wojtossfm ah yeah. We haven't released in some time. I assumed you where working with the latest version from git.

creolis, 2015-10-26:
:) funny thing: I could remember this "bug" somehow, but I didn't know if we already fixed it or just forgot about it ..

wojtossfm, 2015-10-27:
I can confirm that this isn't present in the 0.4 release anymore. Apparently something changed between the 0.4 that I had and the 0.4 that is currently available. Sorry for the mix up and I'm glad that this has been fixed. When I looked at a diff between the main files it was probably related to forking but I didn't delve deeper into it since it's fixed.

katkad, 2015-10-27:
well, there is 168 commits between 0.4.0 and current master :)

wojtossfm, 2015-10-27:
I partially tested the current master yesterday and also didn't notice the issue however I had one of the dependencies in an older version than theoretically required so best if someone reproduces the test with all of the dependencies.

adepasquale, 2015-10-27:
`git bisect` is your friend :smile:

glaslos, 2015-10-27:
Closing this as a know issue in 0.4.0 and fixed in master.

closed 2015-10-27 by glaslos

handsome113530, 2020-04-14:
hello，I  install conpot as required,but  can not open localhost, and namp is normally
[image]
[image]
[image]