mushorg/conpot #239: Error while adding proxy for getting dynamic pages

ramsandesh, 2015-04-27:
For adding dynamic pages to the conpot. I tried the following.
Procedure 1: 
Adding proxy to the status 404 but showed an error. 
<status name="404">
                
                <tarpit>0</tarpit>
		<proxy>127.0.0.1:8080</proxy>
                <entity name="Content-Type">text/html</entity>
 </status>
Error i got : Element 'proxy': This element is not expected. Expected is ( entity ).

Procedure 2 : I even tried adding proxy to node. as below
 <node name="/login_submit.php">
	 
	   <proxy>127.0.0.1:8080</proxy>
</node>

Error i got : 
/usr/local/lib/python2.7/dist-packages/Conpot-0.4.0-py2.7.egg/conpot/templates/default/http/http.xml:53:0:ERROR:SCHEMASV:SCHEMAV_ELEMENT_CONTENT: Element 'proxy': This element is not expected. Expected is one of ( status, tarpit, triggers, headers, alias ).

How can i fix this ? Thanks in advance.

creolis, 2015-04-27:
XSD Schema malformed ... :-/ ... I take a look at it

creolis, 2015-04-27:
sorry for the inconvenience. Please pull from master and try again.

ramsandesh, 2015-04-28:
Thanks.. Error is fixed now. Working fine.

closed 2015-04-28 by glaslos