mushorg/conpot #241: Conpot nobody/nogroup privileges

mav3rick1990, 2015-04-30:
Hello!

I have created a website that I would like to host using conpot's webserver. I have set permissions for all files to be read appropriately. However when I start conpot it drops all privileges. I am afraid this is the reason why the webserver is not reading my CSS and other supporting files on the website. This website is fully functional and works on a normal apache webserver.

Is there a workaround through this? 

Thank you
[image]

glaslos, 2015-04-30:
Do you see any error messages on stdout (console)?
What is the response code on the missing files (check your web browsers development tools (F12))?
Have you compared the permissions on your files with those in the default template?

creolis, 2015-04-30:
Please post your http configuration xml, I am pretty sure I know what's going on :)

creolis, 2015-04-30:
(at least my crystal ball yells "MIME TYPE!!!") .. but we'll see from your xml :).
Btw.: You can also post the XML to something like pastebin and link it here if that's more comfortable for ya.

mav3rick1990, 2015-05-01:
Hi Daniel,

http.xml from my deployment is attached below:

<http enabled="True" host="0.0.0.0" port="80">
    <global>
        <config>
            
            <entity name="protocol_version">HTTP/1.1</entity>
            
            <entity name="update_header_date">true</entity>
            
            <entity name="disable_method_head">false</entity>
            
            <entity name="disable_method_trace">false</entity>
            
            <entity name="disable_method_options">false</entity>
            
            <entity name="tarpit">0</entity>
        </config>

        
        <headers>
            
            <entity name="Date">Sat, 28 Apr 1984 07:30:00 GMT</entity>
        </headers>
    </global>

    
    <htdocs>
        <node name="/">
            
            <status>302</status>
            <headers>
                
                <entity name="Content-Type">text/html</entity>
                <entity name="Location">/index.html</entity>
            </headers>
        </node>
        <node name="/index.html">
            
            <tarpit>0.0;0.3</tarpit>
            <headers>
                <entity name="Last-Modified">Tue, 19 May 1993 09:00:00 GMT</entity>
                <entity name="Content-Type">text/html</entity>
                <entity name="Set-cookie">path=/</entity>
            </headers>
        </node>
        <node name="/index.htm">
            
            <alias>/index.html</alias>
        </node>

        
        <node name="/tests/unittest_base.html">
            <headers>
                <entity name="Last-Modified">Tue, 19 May 1993 09:00:00 GMT</entity>
                <entity name="Content-Type">text/html</entity>
            </headers>
        </node>
        <node name="/tests/unittest_databus.html">
            <headers>
                <entity name="Last-Modified">Tue, 19 May 1993 09:00:00 GMT</entity>
                <entity name="Content-Type">text/html</entity>
            </headers>
        </node>
        <node name="/tests/unittest_tarpit.html">
            <tarpit>5</tarpit>
            <headers>
                <entity name="Last-Modified">Tue, 19 May 1993 09:00:00 GMT</entity>
                <entity name="Content-Type">text/html</entity>
            </headers>
        </node>
        <node name="/tests/unittest_subselects.html">
            <triggers>
                <entity appendix="5459fa05e5c1db37f2679b65a5175bcf">action=unit;subaction=test</entity>
            </triggers>
            <headers>
                <entity name="Last-Modified">Tue, 19 May 1993 09:00:00 GMT</entity>
                <entity name="Content-Type">text/html</entity>
            </headers>
        </node>
    </htdocs>

    
    <statuscodes>
        <status name="400">
                
                <tarpit>0</tarpit>
                <entity name="Content-Type">text/html</entity>
        </status>
        <status name="404">
                
                <tarpit>0</tarpit>
                <entity name="Content-Type">text/html</entity>
        </status>
        <status name="501">
                
                <tarpit>0</tarpit>
                <entity name="Content-Type">text/html</entity>
        </status>
    </statuscodes>
</http>

mav3rick1990, 2015-05-01:
Hi Daniel,

As requested.

<http enabled="True" host="0.0.0.0" port="80">
    <global>
        <config>
            
            <entity name="protocol_version">HTTP/1.1</entity>
            
            <entity name="update_header_date">true</entity>
            
            <entity name="disable_method_head">false</entity>
            
            <entity name="disable_method_trace">false</entity>
            
            <entity name="disable_method_options">false</entity>
            
            <entity name="tarpit">0</entity>
        </config>

        
        <headers>
            
            <entity name="Date">Sat, 28 Apr 1984 07:30:00 GMT</entity>
        </headers>
    </global>

    
    <htdocs>
        <node name="/">
            
            <status>302</status>
            <headers>
                
                <entity name="Content-Type">text/html</entity>
                <entity name="Location">/index.html</entity>
            </headers>
        </node>
        <node name="/index.html">
            
            <tarpit>0.0;0.3</tarpit>
            <headers>
                <entity name="Last-Modified">Tue, 19 May 1993 09:00:00 GMT</entity>
                <entity name="Content-Type">text/html</entity>
                <entity name="Set-cookie">path=/</entity>
            </headers>
        </node>
        <node name="/index.htm">
            
            <alias>/index.html</alias>
        </node>

        
        <node name="/tests/unittest_base.html">
            <headers>
                <entity name="Last-Modified">Tue, 19 May 1993 09:00:00 GMT</entity>
                <entity name="Content-Type">text/html</entity>
            </headers>
        </node>
        <node name="/tests/unittest_databus.html">
            <headers>
                <entity name="Last-Modified">Tue, 19 May 1993 09:00:00 GMT</entity>
                <entity name="Content-Type">text/html</entity>
            </headers>
        </node>
        <node name="/tests/unittest_tarpit.html">
            <tarpit>5</tarpit>
            <headers>
                <entity name="Last-Modified">Tue, 19 May 1993 09:00:00 GMT</entity>
                <entity name="Content-Type">text/html</entity>
            </headers>
        </node>
        <node name="/tests/unittest_subselects.html">
            <triggers>
                <entity appendix="5459fa05e5c1db37f2679b65a5175bcf">action=unit;subaction=test</entity>
            </triggers>
            <headers>
                <entity name="Last-Modified">Tue, 19 May 1993 09:00:00 GMT</entity>
                <entity name="Content-Type">text/html</entity>
            </headers>
        </node>
    </htdocs>

    
    <statuscodes>
        <status name="400">
                
                <tarpit>0</tarpit>
                <entity name="Content-Type">text/html</entity>
        </status>
        <status name="404">
                
                <tarpit>0</tarpit>
                <entity name="Content-Type">text/html</entity>
        </status>
        <status name="501">
                
                <tarpit>0</tarpit>
                <entity name="Content-Type">text/html</entity>
        </status>
    </statuscodes>
</http>

mav3rick1990, 2015-05-01:
Hi Daniel,

As requested.
http://pastebin.com/c9Lb1BAe

mav3rick1990, 2015-05-01:
Hi Lukas,

Thank you for helping me debug the problem. The answer to your questions:
No, I do not have an error conpot runs as per normal. I can see HTTP request and response messages in the logs.  

[image]

This is the error i get in my web browser when I click enter on index.html. I understand that it cannot access the CSS file and other supporting files, hence showing this message. 
[image]

creolis, 2015-05-01:
Hey mav, .. 

your problem is a actually a simple one. You did not customize your xml .. it is totally unconfigured and "default" - so conpot delivers every item that is preconfigured (a few unit tests and index.html) but that is it. It does not know of any other pages or css files.

The webserver is highly customizeable, at the price of having to customize it a lot :)
You need to specify every single file you want to deliver.

creolis, 2015-05-01:
if you want to automate the creation of the node entries, I hacked together a few lines of code that may support you. Just run it from within the htdocs directory, it will take care of a bunch of mime types automatically (and you can easily extend/modify it) and writes the node records to stdout.

Just put them into your configuration xml instead of the nodes you already got there :)

```
import os

for path,dirs,files in os.walk('.'):
    pathname = path[1:]
    if len(pathname) == 0:
        pathname = '/' + pathname
    for filename in files:
        print '<node name="{0}">'.format(os.path.join(pathname, filename))
        print '    <headers>'
        print '        <entity name="Last-Modified">Tue, 19 May 1993 09:00:00 GMT</entity>'

        if filename[-4:].lower() in ['html', 'mwsl'] or filename[-3:].lower() == 'htm':
            print '        <entity name="Content-Type">text/html</entity>'
        if filename[-3:].lower() == 'css':
            print '        <entity name="Content-Type">text/css</entity>'
        if filename[-3:].lower() == 'gif':
            print '        <entity name="Content-Type">image/gif</entity>'
        if filename[-3:].lower() == 'png':
            print '        <entity name="Content-Type">image/png</entity>'
        if filename[-3:].lower() == 'jpg' or filename[-4:].lower() == 'jpeg':
            print '        <entity name="Content-Type">image/jpeg</entity>'
        if filename[-2:].lower() == 'js':
            print '        <entity name="Content-Type">text/javascript</entity>'

        print '    </headers>'
        print '</node>'
```

mav3rick1990, 2015-05-02:
Hi Daniel,

Thanks for pointing me to the right direction. Really appreciate it! I managed to change the http.xml file and included the appropriate nodes. The website is functional now however it appears to me that it still does not recognise the CSS file and other images the website is supposed to show and shows when I run it on a normal webserver. Posting some screenshots below and the updated http.xml file for your perusal.

http://pastebin.com/tQDg7Q8B
[image]
(works perfectly fine. no image corruption)
[image]
(this is the page without css or images included)

Hope to find some direction with this. Thank you!

creolis, 2015-05-02:
Weird ... but sorry, it works for me (tried it with a S7-1200 gif logo and the xml-node from your http xml).
Do you run on latest git master, or did you install using pip?

mav3rick1990, 2015-05-02:
I'm actually running the latest git master. The error doesn't really make any sense. I thought it would be something to do with file permissions, but changing them didn't help either!

glaslos, 2015-05-02:
Have you tried to refresh your browser cache? Ctrl+F5 ☺

mav3rick1990, 2015-05-02:
Yes! In fact I got rid of my cache and other history. I ensured that it is a clean slate before accessing local host online.

creolis, 2015-05-02:
what I can offer you is that you send me a zip containing your template as a whole and I will run it in a test environment here to see if it misbehaves the same way you experience..

mav3rick1990, 2015-05-03:
Hi Daniel,

I have uploaded the entire folder online. You can find it at the below mentioned address:
http://www.4shared.com/zip/gEMUXnijba/web_root.html

Thanks for your help!

creolis, 2015-05-03:
err, I'm not going to make an account at 4shared (or use social auth) :)
Please upload via FTP here:
91.219.68.113

since we got no PM here at github, pls. provide me with your eMail address, so that I can send you the authentication details.

mav3rick1990, 2015-05-03:
Hi Daniel,

Sure. I am game for any which way the problem gets solved. Here you go;
onion.logmanager@gmail.com. 

Thanks a lot!

mav3rick1990, 2015-05-04:
Hi Daniel,

I have uploaded the data on your FTP server. Just wanted to let you know :)

Thank you for your help!

mav3rick1990, 2015-05-04:
Hi Daniel,

I re-installed conpot from scratch and edited the xml file to include new nodes. Now, to my surprise it is working! Can't seem to diagnose the problem but whatever it was, got fixed. I just wanted to let you know that its in place now. I can finally access CSS/other files.

Thank you for your help and support on this matter! Really helped me a lot. :)

This is a good project and I hope there is development support available for this for a long time. 

Cheers!

creolis, 2015-05-04:
Glad to hear it's working now :) 
If you have further questions, don't hesitate to get in touch with us. We also accept enhancements if you happen to change parts of the code :)

closed 2015-05-04 by creolis