mushorg/conpot #242: Problem in post request handling by conpot web server

ramsandesh, 2015-05-03:
I have made a customized HMI. I have implemented authentication for the HMI. For loading the Dynamic pages i am using proxy, which is running on localhost(127.0.0.1, 8080). 
Redirecting is working fine, but have a problem while authenticating. I am using php code. When the  POSTrequest comes from localhost to conpot, it is unable to process. I am not sure how to rectify this problem. please find the attachment. 

[image]

creolis, 2015-05-03:
*heh* I think I know exactly what is happening ...

okay, so this is a case that is not really handled yet .. I looked at the code and this edge case should definitely be enhanced.

You send a POST request - the request is trying to get a file via configuration but could not find a valid configuration item, so it's a non-existent file and therefore a 404. A 404 status is generated, but due do configuration, it is not fetched from disk but from a remote system (using the proxy statement). This will issue a GET request. So basically we convert POST to GET in proxied status page scenarios... and this results in the trouble you're seeing.

I'll try to find some spare time and fix this. How time critical is this issue for you?

ramsandesh, 2015-05-03:
I am stuck with this from 3-4 days, i was thinking it is my installation issues, so i was reinstalling everything and wasted so much time. Take your time, but i need as soon as possible.

creolis, 2015-05-06:
ramsandesh:
I just gave it a quick shot without any testing and added a few things into the blue.
If you want, you may try if this works for you instead of waiting until the change has been tested and made it into master:

https://github.com/glastopf/conpot/tree/proxy_method_awareness

ramsandesh, 2015-05-08:
Sorry it didn't work. Please have a look at the screen shot attached.

[image]

ramsandesh, 2015-05-14:
Can any one help me in fixing this ??

creolis, 2015-05-14:
sorry, I was quite busy this week and I did not forget you.

Of course you can have a look at the initial changes I made in the proxy method awareness branch and see what I might have missed - you are welcome to send in corrections at any time :)

creolis, 2015-05-21:
Did you find out anything yourself yet?

I had a look at traffic dumps, using a direct connection and using a conpot proxied connection.
The conpot proxied connection did not successfully transport POST variables - but I can't tell why because both requests are basically the same. I'm not sure what's wrong here .. 

ideas anyone?

[image]

creolis, 2015-05-21:
content type maybe ... *hmm*

creolis, 2015-05-21:
yeah .. verified it via plain ol' telnet. A content type header seems to be mandatory for further processing of the request body. So basically, what we have to do here is stuffing all those headers originated by the client to the subsequent request ... with some modifications though as we will need to replace the Host: and Connection: headers.

creolis, 2015-05-25:
@ramsandesh Please retry the proxy awareness branch again. I made changes that tackle the POST issue while at the same time fixing the fact that proxied connections did not handle cookies correctly - which they do now too.

I tested it and it seems that everything works fine so far, but I'll wait for your confirmation before I merge.

ramsandesh, 2015-05-25:
Thanks a lot. Let me try the changes. will update you in 12hours.

creolis, 2015-05-25:
kk :)

ramsandesh, 2015-05-26:
Its working. Perfect (y). Thanks a lot. I tested it, Its handling the POST requests perfectly.

creolis, 2015-05-26:
Great :) sorry that it took so long, $dayjobs can be pretty time consuming :-)

closed 2015-05-26 by creolis