mushorg/conpot #243: Nmap Scan Results

mav3rick1990, 2015-05-06:
Hello,

I have my conpot machine setup for exploitation. However, when I tried to run an nmap scan over it, the only open port i see is port 80. In actual case, 102,502 and 161 are open too. I am running modbusdetect module from metasploit to identify a modbus service running.

Is  this supposed to be normal behaviour or am I missing something?
[image]

glaslos, 2015-05-06:
Can you do a `netstat -lnptu`?

mav3rick1990, 2015-05-06:
[image]
Output of the command on conpot machine

Vingaard, 2015-05-06:
quick question, do you remember to include UDP in the scan,  like  nmap -sU

mav3rick1990, 2015-05-06:
Hi,
I am not getting any results from UDP scans. TCP scan sort of dumps an error in the conpot.log as well. SYN scan results in showing that only port 80 is open. 'A' flag also just shows that port 80 is open. Only when I specify -p 502 or other port numbers, does it show that other ports are open too.

I hope I have explained the problem I face.

Thank you for helping!

creolis, 2015-05-06:
... please send us the exact nmap command line you are using to start the scan.

creolis, 2015-05-06:
Uhm .. from your nmap-screenshot I can see the following line:
"Not shown: 999 closed ports".

This means that you obviously ran nmap pretty much unconfigured, did not tell it what to look at but just used the default profile, (maybe as SYN- or CONNECT scan).

What you should know using nmap is that running it unconfigured means that it will only test ~ 1000 of the most common ports it knows, _not every port that's possible_. And 102, 502 etc. are "well known", but not very common in the wild.

So there are two things to consider:
1) start nmap using the ports directive: nmap -sS -p1-1000, this will SYN-Scan all ports in the range from 1 - 1000, .. a full scan would be -p1-65535 and definitely take its time
2) know your tools by heart, .. that comes in handy :)

closed 2015-05-06 by creolis

creolis, 2015-05-06:
( oh .. and since we are educating here - UDP scans are a pain in the ass as you don't scan for existence but for NON-existence of open ports. UDP scans rely on the fact that probes to closed ports are answered using ICMP messages, but this is troublesome since operating systems decided to limit the number of ICMP messages per second (ICMP-rate-limiting). If you scan too fast, you will simply MISS some messages, which means that your scan is not trustworthy. On linux, the rate limit for certain types of ICMP messages is 1000ms between two responses, so a proper UDP scan will use like 1 probe per second, which translates to ~ 11 hours of scanning to get reliable results for the whole UDP port range.

mav3rick1990, 2015-05-06:
Hi Daniel,

'And 102, 502 etc. are "well known", but not very common in the wild.'

Definitely got to learn something new here. I know how to use nmap just not enough experience handling these protocols. Thank you for your guidance!

Cheers!

creolis, 2015-05-06:
... yeah, it's a fun thing - when you look at:
https://nmap.org/book/man-port-specification.html

it says: "By default, Nmap scans the most common 1,000 ports for each protocol".

But up to now I did not find a list of what those 1000 ports actually ARE. The reason I knew this by heart is because I ran into the very same issues a lot of times while doing penetration tests.