mushorg/conpot #246: Assertion error in Kamstrup

glaslos, 2015-05-09:
```Python
Traceback (most recent call last):
  File "/home/travis/virtualenv/python2.7.9/lib/python2.7/site-packages/gevent-1.0.1-py2.7-linux-x86_64.egg/gevent/greenlet.py", line 327, in run
    result = self._run(*self.args, **self.kwargs)
  File "/home/travis/build/glastopf/conpot/conpot/protocols/kamstrup/usage_simulator.py", line 54, in initialize
    self.energy_in = databus.get_value(energy_in_register)
  File "/home/travis/build/glastopf/conpot/conpot/core/databus.py", line 44, in get_value
    assert key in self._data
AssertionError
<Greenlet at 0x7feb29971e10: <bound method UsageSimulator.initialize of <conpot.protocols.kamstrup.usage_simulator.UsageSimulator object at 0x7feb29413b10>>> failed with AssertionError
```

johnnykv, 2015-05-09:
When do you get this? At startup?

glaslos, 2015-05-09:
In the Travis test

johnnykv, 2015-05-09:
Yeah ok, I see it. I bet it is some kind of closedown logic gone wrong.

commit 5dede9cf (2015-05-09): closedown flow for instances used as value in databus

fix #246
closed 2015-05-09 by johnnykv