mushorg/conpot #251: Create documentation on how to use Docker

glaslos, 2015-05-12:
There should be some bare minimum instructions on how to use docker to run Conpot.

adepasquale, 2015-07-22:
I will write some documentation based on [this comment](https://github.com/mushorg/conpot/pull/201#issuecomment-75657640) from issue #201.

xandfury, 2017-01-28:
**How is this for a start?** 
 
EASY INSTALL USING DOCKER CONTAINER
******************************************************
1. Install `Docker <https://www.docker.com/>`_
2. Run ``docker pull honeynet/conpot``
3. Run ``docker run -it -p 80:80 -p 102:102 -p 502:502 -p 161:161/udp --network=bridge honeynet/conpot:latest /bin/sh``
4. Finally run ``conpot --template default``

Navigate to ``http://MY_IP_ADDRESS`` to confirm the setup.

PR #318 (2017-02-22): Easy Docker Container Installation Instructions
Add documentation for docker install using this commit.
Fix #251
closed 2017-02-22 by glaslos