mushorg/conpot #261: [bacnet] DecodingError: invalid APDU type

TanyaTie, 2015-06-21:
I am work on ICS honeypot using bacnet protocol in conpot.
I setup conpot, when I test bacnet using nmap (nmap --script bacnet-info -sU -p 47808 ipaddr),
got follow errors:
Traceback (most recent call last):
  File "/usr/lib64/python2.6/site-packages/gevent-1.1-py2.6-linux-x86_64.egg/gevent/greenlet.py", line 340, in run
    result = self._run(*self.args, **self.kwargs)
  File "/usr/lib64/python2.6/site-packages/gevent-1.1-py2.6-linux-x86_64.egg/gevent/baseserver.py", line 132, in _close_when_done
    return handle(*args)
  File "/usr/lib/python2.6/site-packages/Conpot-0.4.0-py2.6.egg/conpot/protocols/bacnet/bacnet_server.py", line 411, in handle
    apdu.decode(pdu)
  File "/usr/lib/python2.6/site-packages/BACpypes-0.12.0-py2.6.egg/bacpypes/apdu.py", line 344, in decode
    APCI.decode(self, pdu)
  File "/usr/lib/python2.6/site-packages/BACpypes-0.12.0-py2.6.egg/bacpypes/apdu.py", line 283, in decode
    raise DecodingError, "invalid APDU type:" + str(self.apduType)
DecodingError: invalid APDU type:8
<Greenlet at 0x4782c30: _close_when_done('\x81\n\x00\x11\x01\x04\x00\x05\x01\x0c\x0c\x02?\x, ('ipaddr', 27157))> failed with DecodingError

actually, APDU type would be 0. I think APDU is checked by wrong start index. How can I fix it?

glaslos, 2015-06-21:
Have you checked if we pass the wrong data to the bacpypes decoder?

TanyaTie, 2015-06-21:
but I use the same namp scan to the internet, can get the result like that:
nmap --script bacnet-info -sU -p 47808 -Pn <ipaddr>

Starting Nmap 6.47 ( http://nmap.org ) at 2015-06-21 21:15
Nmap scan report for <domain> (ipaddr)
Host is up.
PORT      STATE SERVICE
47808/udp open  bacnet
| bacnet-info:
|   Vendor ID: Reliable Controls Corporation (35)
|   Vendor Name: Reliable Controls Corporation
|   Object-identifier: 1000
|   Firmware: 7.70
|   Application Software: 7.70
|   Object Name: Q134-P1
|   Model Name: MACH-ProCom
|   Description:
|_  Location:

Nmap done: 1 IP address (1 host up) scanned in 4.86 seconds

I also use whois command of "bacnet tools", the APDU Type is "8" too.

glaslos, 2015-06-21:
So for some reason bacpypes can't handle the type. I'll try to have a look next week.

katkad, 2015-06-23:
Hi, I had commit in testing which fixed this exception
@TanyaTie thank you for reproducer

commit is now in master https://github.com/glastopf/conpot/commit/7a9b0e961b47e2918cc27e9c281531a7b7f5d641, and logs now look like this:
2015-06-23 12:56:43,929 DecodingError: invalid APDU type:8
2015-06-23 12:56:43,930 PDU: <PDU None -> None : 0A.00.11.01.04.00.05.01.0C.0C.02.3F.FF.FF.19.4B>

so you can see what is sent.
It corresponds to function send_query in https://svn.nmap.org/nmap/scripts/bacnet-info.nse

This commit doesn't resolve APDU Type issue though, just is usefull for debugging it.

TanyaTie, 2015-06-26:
@katkad
Thank you very mach. I will look the debug info.

glaslos, 2015-11-08:
@TanyaTie did you had a chance to look into this?

TanyaTie, 2015-11-09:
@glaslos 
I found out that origin python source file read wrong place to decode the APDU Type. So I analysis the pcap file by wireshark, and refer to analysis.py file in bacpypes. And I finally decode the data. So far, I only get the ReadProperty Service run, it's  hard to understand whole bacpypes.

glaslos, 2015-11-09:
Could you elaborate a bit on your findings?

TanyaTie, 2015-11-11:
@glaslos 
First, I add a method --"decodeBacnetUdp" in class BacnetServer. This method will decode the udp data from "handle" method. Method "decodeBacnetUdp" refer to  decode_packet(data) method of analysis.py file in bacpypes, which can return a decoded apdu data.
And then, in handle method of class BacnetServer, I use the decoded apdu to do "bacnet_app.indication" and "bacnet_app.response". However, the bacnet services defined in bacnet_app.py file should be modified a little. I have modified Whois-IAM service and readProperty service. I am still working on other services.

windflyer, 2017-07-11:
This is a decoding bug in BACpypes, see here: https://github.com/JoelBender/bacpypes/issues/105

Upgrading to BACpypes 0.16.1 solves this issue.

creolis, 2017-07-11:
Fixed with PR #335

closed 2017-07-11 by creolis

Rajeswara-Muthyala, 2021-11-03:
@yurushao @TanyaTie I still see the issue on the latest version. It appears to me that conpot code needs an change here. 

WARNING:conpot.protocols.bacnet.bacnet_server:DecodingError - PDU: <PDU None -> None : 0a.00.11.01.04.00.05.01.0c.0c.02.3f.ff.ff.19.4b>
2021-11-03 10:46:44,151 DecodingError - PDU: <PDU None -> None : 0a.00.11.01.04.00.05.01.0c.0c.02.3f.ff.ff.19.4b>

The above error is with latest conpot version. requirements.txt points to bacpypes==0.17.0