mushorg/conpot #262: conpot error with downloaded files with hmi_crawler

arunm2110, 2015-06-22:
I am tring to give new hmi look to conpot with hmi_crawler
i downloaded hmi with hmi_crawler, when i am using commands with conpot like,
conpot -w <your new directory> -r <root page> , it shows unrecognized arguments.

1) 
a-virtual-machine ~ # conpot -w ~/ics/dump/ -r ~/ics/dump/Portal4300.htm 

                       _
   ___ ___ ___ ___ ___| |_
  |  _| . |   | . | . |  _|
  |___|___|_|_|  _|___|_|
              |_|

  Version 0.4.0
  Glastopf Project

usage: conpot [-h] [-t TEMPLATE] [-c config.cfg] [-l LOGFILE] [-a RAW_MIB]
              [-m MIBPATHS] [-v]
conpot: error: unrecognized arguments: -w /root/ics/dump/ -r /root/ics/dump/Portal4300.htm

2)
a-virtual-machine ~ # conpot -w /dump2 -r Default.html

                       _
   ___ ___ ___ ___ ___| |_
  |  _| . |   | . | . |  _|
  |___|___|_|_|  _|___|_|
              |_|

  Version 0.4.0
  Glastopf Project

usage: conpot [-h] [-t TEMPLATE] [-c config.cfg] [-l LOGFILE] [-a RAW_MIB]
              [-m MIBPATHS] [-v]
conpot: error: unrecognized arguments: -w /dump2 -r Default.html

3)
a-virtual-machine ~ # conpot -w ~/ics/dump2/ -r Default.html

                       _
   ___ ___ ___ ___ ___| |_
  |  _| . |   | . | . |  _|
  |___|___|_|_|  _|___|_|
              |_|

  Version 0.4.0
  Glastopf Project

usage: conpot [-h] [-t TEMPLATE] [-c config.cfg] [-l LOGFILE] [-a RAW_MIB]
              [-m MIBPATHS] [-v]
conpot: error: unrecognized arguments: -w /root/ics/dump2/ -r Default.html

4)
a-virtual-machine ~ # conpot -w ~/ics/dump/ -r index.html

                       _
   ___ ___ ___ ___ ___| |_
  |  _| . |   | . | . |  _|
  |___|___|_|_|  _|___|_|
              |_|

  Version 0.4.0
  Glastopf Project

usage: conpot [-h] [-t TEMPLATE] [-c config.cfg] [-l LOGFILE] [-a RAW_MIB]
              [-m MIBPATHS] [-v]
conpot: error: unrecognized arguments: -w /root/ics/dump/ -r index.html

Above are attempts i did , but having same issues.

glaslos, 2015-07-14:
The HMI document (https://github.com/mushorg/conpot/blob/master/docs/source/usage/hmi.rst) is outdated. Have a look at https://github.com/mushorg/conpot/blob/master/docs/source/usage/usage.rst#http

arunm2110, 2015-07-16:
Thank you so much for help

closed 2015-07-16 by glaslos