mushorg/conpot #266: Centos Support on Conpot

Vingaard, 2015-07-08:
As a follow-up to a previous issue (#230) - now closed, I commission a freelancer (who failed) to make conpot work on Centos. After some additional troubleshooting time, I found below method to install conpot on Centos.

I would kindly ask interested peers to test below, and provide feedback. Should the community feel the guideline are correct/sufficient, I will make a pull request. I am not a centos expert but this has been validated to work on Centos version 7.1-1503 (but may likely work on other centos versions/ rpm disto.

#1 login via ssh with a account with sufficient system privileges (e.g root)
#2 Upgrade the system
sudo yum -y update
#3 Install epel repository
sudo rpm -iUvh http://dl.fedoraproject.org/pub/epel/7/x86_64/e/epel-release-7-5.noarch.rpm
#4 Install needed packages and libs
sudo yum -y install libxslt-devel libxml2-devel python-pip python-2.7.5-16.el7.x86_64 mariadb-server mysql-connector-python.noarch mariadb-devel-5.5.41-2.el7_0.x86_64 git python-lxml.x86_64 python-devel
sudo yum -y groupinstall "Development tools"
sudo easy_install -U setuptools
#force lxml to be version 3.3.5
sudo easy_install lxml==3.3.5
#5 starting mysql server
sudo chkconfig mariadb on
sudo service mariadb start
sudo mysql_secure_installation

#6 CONPOT installation
cd /usr/local/src
sudo git clone https://github.com/glastopf/conpot
cd conpot/
sudo python setup.py install

#7 open ports in firewalld : 80 , 102, 161 and 502
firewall-cmd --permanent --add-port=80/tcp
firewall-cmd --permanent --add-port=102/tcp
firewall-cmd --permanent --add-port=161/tcp
firewall-cmd --permanent --add-port=502/tcp
firewall-cmd --reload

#8 check if its running and you can access it from remote :
in browser
http://YOUR_IP/

I am looking forward to hear your comments.

adepasquale, 2015-07-09:
Thanks for sending your script, I have just tested it and here are some notes.

* The number of new packages installed is huge (roughly 100-150), are you sure all of them are absolutely needed? What is pulled in with `sudo yum -y groupinstall "Development tools"`?
* I'd add some suggestions about `mysql_secure_installation` questions, i.e. what to answer for a production environment.
* I'd also consider cloning a stable version of conpot instead of master, or maybe using [the one from PyPI](https://pypi.python.org/pypi/Conpot/).
* There are some warnings when building some of the python dependencies, but one should not worry about that.
* A couple of lines about how to configure and start conpot could also be added, e.g. `conpot --template default` (and related firewalld config).
* There is an exception, which the conpot team will need to investigate, when conpot tries to drop privileges:
```
Traceback (most recent call last):
  File "/usr/bin/conpot", line 4, in <module>
    __import__('pkg_resources').run_script('Conpot==0.4.0', 'conpot')
  File "build/bdist.linux-x86_64/egg/pkg_resources/__init__.py", line 735, in run_script
  File "build/bdist.linux-x86_64/egg/pkg_resources/__init__.py", line 1652, in run_script
  File "/usr/lib/python2.7/site-packages/Conpot-0.4.0-py2.7.egg/EGG-INFO/scripts/conpot", line 356, in <module>
    main()
  File "/usr/lib/python2.7/site-packages/Conpot-0.4.0-py2.7.egg/EGG-INFO/scripts/conpot", line 332, in main
    drop_privileges()
  File "/usr/lib/python2.7/site-packages/Conpot-0.4.0-py2.7.egg/EGG-INFO/scripts/conpot", line 104, in drop_privileges
    wanted_gid = grp.getgrnam(gid_name)[2]
KeyError: 'getgrnam(): name not found: nogroup'
```

In the end, these instructions work good, but need some minor fixes. If you make a pull request, could you please submit the instructions as a [centos.rst documentation file](https://github.com/mushorg/conpot/tree/master/docs/source/installation)?

glaslos, 2015-07-09:
Thanks for testing Andrea. 'nogroup' is probably Debian specific and we have to figure out what we should drop to on CentOS

Vingaard, 2015-07-14:
Hello adepasquale , first - many thanks for your time and input, I will fix the items and submit  it as a centos.rst. After initial submit, I will go to the lab and see if I can cut down on the packages required

adepasquale, 2015-07-14:
Thank **you** @Vingaard for sharing your instruction with the community. :smile: 

Looking forward to your pull request!

Vingaard, 2015-07-20:
You are most welcome - many thanks for your input. Updated rst file send as PR

closed 2015-07-20 by Vingaard