mushorg/conpot #269: ImportError: No module named bacnet_server

ahmedsherif, 2015-07-13:
I tried to install conpot recently and I figured out this error while executing conpot 

"from conpot.protocols.bacnet.bacnet_server import BacnetServer
ImportError: No module named bacnet_server" 

When I tried to get in to Modulels "/conpot/conpot/protocols/bacnet" I couldn't get the bacnet_server

glaslos, 2015-07-13:
This is a known issue in the current master branch.

johnnykv, 2015-07-13:
Are you getting this from current? Afaik that should be fixed a few nights ago (But IPMI still fails).

glaslos, 2015-07-13:
Johnny: 
https://github.com/mushorg/conpot/blob/master/bin/conpot#L47
https://github.com/mushorg/conpot/blob/master/conpot/protocols/bacnet/bacnet_app.py

bacnet_app vs. bacnet_server

johnnykv, 2015-07-14:
fixed with 1da8580c152b7b6443c174382cfe501e940f0744

closed 2015-07-14 by johnnykv