mushorg/conpot #272: IPMI server or broken

johnnykv, 2015-07-17:
I Properly broke this. For details see threatstream/mhn#185

glaslos, 2015-07-17:
MHN installs Conpot from master?

glaslos, 2015-07-17:
Maybe we should have some more stable, tagged releases :)

katkad, 2015-07-17:
@glaslos good idea, but what I _really_ would like is working master at all time

glaslos, 2015-07-17:
@katkad I agree with you too. We are very sloppy when it comes to QA and product management.

adepasquale, 2015-07-17:
Development branch?

glaslos, 2015-07-17:
I think we did that with Glastopf at one point. Problem is, we then need reliable releases to the master branch so it doesn't lag too far behind. Otherwise you end up with replying to every issue with 'Check out the development branch'.

Spindel, 2015-07-17:
Can I suggest a time based release schedule then?

johnnykv, 2015-07-17:
We also need better test coverage, currently all tests pass.

katkad, 2015-08-12:
not broken master

closed 2015-08-12 by katkad