mushorg/conpot #280: Still having MAC masquerading issues...

infracritical, 2015-09-09:
Trying to work with either Ubuntu 14.04 or 14.10, either one patched, too.

For the life of me, I *cannot* seem to get the MAC masquerading working.  I've tried it with it 'on' with the 'hwaddress' field left out, and with it 'off' with the 'hwaddress' field added in.  In either circumstance, when I execute NMAP, it comes back as being a Linux O/S -- actually, I believe that it says 'Mandriva'.

Can someone help me with this?  I've gotten this to mysteriously to work on other servers running, coincidentally, 14.10.  But somehow, it doesn't seem to want to work -- either with the release that I had downloaded back in April 2015, or recently.

Any ideas, guys???

glaslos, 2015-09-15:
Are you trying to change the IP address? Can you confirm that the interfaces hardware address has changed?

Vingaard, 2016-09-05:
Hello, a workaround (outside Conpot) could be the following commands on ubuntu; (replace eth0 if required) 
ifconfig eth0 down hw ether <MAC_ADDRESS>
ifconfig eth0 up
ifconfig -a

closed 2026-09-14 by glaslos