mushorg/conpot #281: Has anyone tested 'conpot' with CentOS v6?

infracritical, 2015-09-10:
Documentation mentions about v7:

https://github.com/mushorg/conpot/blob/master/docs/source/installation/centos.rst

Just curious about this.  Honestly, I don't trust v7...not...just...yet.  Anyone got any clarification on the CentOS install?  If at all, I could be a guinea pig here, if no one knows...

glaslos, 2015-09-10:
@Vingaard do you have some spare time to do the test you did for v7 also for v6? Best case would be if you just had to swap out the EPEL source.

infracritical, 2015-09-10:
Actually, I *think* that I've got it working on CentOS 6.7; however, it is *much* more complex than what was outlined for CentOS 7.x.  So far, I've managed to get the instance up and running, but somehow cannot connect to the environment.  It's on an isolated server behind 2 firewalls, so I'm checking there for other issues.

BTW, it's more than simply swapping out the EPEL source... :(

infracritical, 2015-09-10:
So far, these are the *raw* steps that I have for the installation.  I need to re-trace my steps on a new/clean server image, then go over each step identified 9I forgot to capture everything into a log file....DOH!!!).

Here are the steps that I have so far...

yum -y update
wget http://dl.fedoraproject.org/pub/epel/6/x86_64/epel-release-6-8.noarch.rpm
rpm -ivh epel-release-6-8.noarch.rpm
yum repolist
yum -y install libxslt-devel libxml2-devel python-pip
yum -y install mariadb-server mysql-connector-python.noarch
yum -y install python-devel
yum -y install python-lxml openssl-devel
yum -y groupinstall "Development tools"
easy_install -U setuptools
wget http://python.org/ftp/python/2.7.5/Python-2.7.5.tar.bz2
tar -xvf Python-2.7.5.tar.bz2
cd Python-2.7.5
./configure --prefix=/usr/local
make -j4
make altinstall
wget http://pypi.python.org/packages/source/d/distribute/distribute-0.6.35.tar.gz
cd distribute-0.6.35
python2.7 setup.py install
easy_install-2.7 pip
pip2.7 install --upgrade pip
pip2.7 install virtualenv
virtualenv /usr/local/python-env --distribute
source /usr/local/python-env/bin/activate
which python
python --version
pip install gunicorn supervisor flask
vi /etc/yum.repos.d 
 -- add
[mariadb]
name = MariaDB
baseurl = http://yum.mariadb.org/5.5/centos6-x86
gpgkey=https://yum.mariadb.org/RPM-GPG-KEY-MariaDB
gpgcheck=1
yum repolist
yum install -y MariaDB-server MariaDB-client MariaDB-devel
source /usr/local/python-env/bin/activate
easy_install lxml
cd /opt
git config --global http.postBuffer 524288000
git clone https://github.com/mushorg/modbus-tk.git
git clone https://github.com/mushorg/modbus-tk.git
cd modbus-tk
python setup.py install
cd ../conpot
python setup.py install
groupadd nogroup

infracritical, 2015-09-10:
Oh, and one more thing...

Edit the /etc/sysconfig/selinux/config file and set "SELINUX=disabled"...

infracritical, 2015-09-10:
HAHZAH!!!!

I *just* rebooted the primary firewall, and IT WORKS!!!

infracritical, 2015-09-10:
I will re-trace my steps again this evening on a new/clean server, and take screenshots, and everything.  I should have a much more detailed manual on the installation process.  I may even attempt this with CentOS 7.x on one of the 64-bit servers that we aren't currently using.

It is much more complicated, but should do the trick!!!.  Again, this was verified on CentOS 6.7 on 32-bit architecture.  I would imagine that it should work comparatively similar for 64-bit architectures, too.

infracritical, 2015-09-10:
I am going to keep this issue open until I can manage to re-produce what I did *cleanly*.

Vingaard, 2015-09-11:
@glaslos Currently going abroad to present the value of running honeypots on a security conference - will be back late next week in my lab as i didn't look at 6.x
@infracritical: one thing who gave me some initial problems was the lack of correct lxml - had to set the version strick to lxml==3.3.5 after lot of troubleshooting, the same may be the case for 6.x.
let me know of progress - would be happy to assist late next week.

infracritical, 2015-09-11:
I was able to verify everything...sort of.  One sentence: it...is...a...mess.

I *think* that I've got the entire process verified; just going to check one more time to make sure that it is verifiable.  This was tested with a baseline load of CentOS 6.6, then upgraded to the most recent, stable release of CentOS (which, as of today, is "6.7").

Once I have it done, I will create a bash script for the installation, and would appreciate it someone from this group were able to review the steps just to make sure that I am not doing too much.

For those reading this issue -- DO NOT -- use the steps outlined above.  There are some steps MISSING.  I am hashing out the details for a CentOS 6.x release process.

More to come...

infracritical, 2015-09-11:
One more thing for those wanting to install this onto a CentOS platform, I would *strongly* recommend using "daemonize".  This will enable the use of the "conpot" product, but placed into a daemon state.

Steps are as follows:

(1)  cd /root
(2)  git clone https://github.com/bmc/daemonize.git
(3)  cd daemonize
(4)  ./configure
(5)  make all
(6)  cp daemonize /usr/local/bin

The final step to executing the "conpot" product is this:

daemonize /usr/local/python-env/bin/conpot -t default

There are a few assumptions:

(1)  That you've been successful with the Python 2.7.5 installation process.
(2)  That you've been successful with the "modbus-tk" and "conpot" installation process.
(3)  That you've been successful with installing the "Development tools" group package.

Also, please ensure that you make your firewall changes, or add into the "/etc/sysconfig/iptables" file.

Also, please ensure that you make SELINUX changes, and change "enforcing" to "disabled".

infracritical, 2015-09-13:
OK...it took a few times to get it straight, but this is a *CLEAN* installation for CentOS 6.x...

This was tested using CentOS 6.6 32-bit architecture as the baseline, then upgraded to CentOS 6.7.

===========================

yum -y update
yum -y install wget git zip unzip
wget http://dl.fedoraproject.org/pub/epel/6/x86_64/epel-release-6-8.noarch.rpm
rpm -ivh epel-release-6-8.noarch.rpm

vi /etc/yum.repos.d/MariaDB.repo

[mariadb]
name = MariaDB
baseurl = http://yum.mariadb.org/5.5/centos6-x86
gpgkey=https://yum.mariadb.org/RPM-GPG-KEY-MariaDB
gpgcheck=1

yum repolist
yum -y install libxslt-devel libxml2-devel openssl-devel
yum -y install python-pip python-devel python-lxml
yum -y install mysql-connector-python.noarch sqlite-devel bzip2-devel
yum -y install MariaDB-server MariaDB-client MariaDB-devel
yum -y groupinstall development

easy_install -U setuptools
wget http://python.org/ftp/python/2.7.5/Python-2.7.5.tar.bz2
tar -xvf Python-2.7.5.tar.bz2
cd Python-2.7.5
./configure --prefix=/usr/local
make -j4
make altinstall

cd /root
wget http://pypi.python.org/packages/source/d/distribute/distribute-0.6.35.tar.gz
tar -xvf distribute-0.6.35.tar.gz
cd /root/distribute-0.6.35
python2.7 setup.py install
cd /root

easy_install-2.7 pip
pip2.7 install --upgrade pip
pip2.7 install lxml

pip2.7 install virtualenv
virtualenv /usr/local/python-env --distribute
###source /usr/local/python-env/bin/activate

cd /opt
git config --global http.postBuffer 524288000
git clone https://github.com/mushorg/modbus-tk.git
cd /opt/modbus-tk
python2.7 setup.py install
cd /opt
git clone https://github.com/mushorg/conpot.git
cd /opt/conpot
python2.7 setup.py install
cd /root
/usr/local/python-env/bin/conpot

git clone https://github.com/bmc/daemonize.git
cd /root/daemonize
./configure
make all
cp daemonize /usr/local/bin
cd /root
daemonize /usr/local/bin/conpot -t default
service iptables stop

http://<IP_ADDR>

infracritical, 2015-09-13:
Somehow the last line got munged.  It should be "http://IP-ADDRESS"...where "IP-ADDRESS" is whatever IP address you've assigned your service.

Also, you may want to add a few entries into your "/etc/sysconfig/iptables" file:

# Firewall configuration written by system-config-securitylevel
# Manual customization of this file is not recommended.
*filter
:INPUT ACCEPT [0:0]
:FORWARD ACCEPT [0:0]
:OUTPUT ACCEPT [0:0]
:RH-Firewall-1-INPUT - [0:0]
-A INPUT -j RH-Firewall-1-INPUT
-A FORWARD -j RH-Firewall-1-INPUT
-A RH-Firewall-1-INPUT -i lo -j ACCEPT
-A RH-Firewall-1-INPUT -m state --state ESTABLISHED,RELATED -j ACCEPT

#-- port [https/http] 443/80
-A RH-Firewall-1-INPUT -p tcp -m tcp --dport 443 -j ACCEPT
-A RH-Firewall-1-INPUT -p tcp -m tcp --dport 80 -j ACCEPT

#-- this assumes that you're located on a "192.168.1.x" subnet; change to whatever you feel;
#-- this is meant to restrict SSH access to ONLY the 192.168.1.x subnet
-A RH-Firewall-1-INPUT -p tcp -m tcp -s 192.168.1.0/24 --dport 22 -j ACCEPT

#-- drop everything else
-A RH-Firewall-1-INPUT -j DROP
COMMIT

infracritical, 2015-09-13:
And that's it!!!  This was tested several times -- FROM SCRATCH -- and has been verified several times.  The environment that this was tested on is a 32-bit virtualized environment, running VMWARE 4.1 Update 3 (4.1U3), running CentOS 6.6, upgraded to its new baseline (CentOS 6.7, as of today -- 13-Sep-2015).

infracritical, 2015-09-14:
Oh...and one more important note...

For the virtualized environment, I used 1 GB of RAM, and 8 GB of disk.  Worked VERY well with this configuration.  If you use anything < 1 GB, it will take considerably slower to install, compile and deploy.

infracritical, 2015-09-14:
I am closing this issue...

closed 2015-09-14 by infracritical