mushorg/conpot #282: Add support for Schweitzer SEL-2032

adepasquale, 2015-10-28:
https://www.selinc.com/SEL-2032/
https://www.ebay.com/sch/i.html?_nkw=Schweitzer+SEL2032

Looks like a telnet plaintext protocol...

infracritical, 2015-11-22:
This device is the equivalent of a remote telemetry unit (RTU), as it supports the DNP3 protocol.  There is *no* Ethernet interface, and must be purchased separately from SEL.  Also, the devices does *not* have a web interface or web admin interface.  This device is a plain 'n simple device -- minimalistic.

infracritical, 2015-11-22:
BTW, this device supports DNP3 and MODBUS.

adepasquale, 2015-11-22:
@infracritical Do you have more info about the telnet server? It should be called SEL-704 or -705.

infracritical, 2015-11-22:
I don't know about the 704 or 705 models.  Are they supposed to be serial->Ethernet converters?

adepasquale, 2015-11-23:
I am sorry @infracritical, I meant [SEL-734](https://www.selinc.com/sel-734/) and [SEL-735](https://www.selinc.com/SEL-735_Revenue_Meter/). Looking at Schweitzer's website they do not look like Ethernet converters, but rather "Power Quality and Revenue Meters".

I've been able to [find some on Shodan](https://www.shodan.io/search?query=sel-+telnet+server), they have a telnet server that _should_ reply to the `access` command. I haven't tested anything yet, for more information please see [this Twitter thread](https://twitter.com/a_de_pasquale/status/661277517974736896).

Given that many different SEL-* devices might reply to the same command could mean there is something in front of them. How about a combination of these?
* SEL-2032 Communications Processor
* SEL-2701 Ethernet Processor Guideform Specification
* SEL-2890 Ethernet Transceiver
* SEL-3610 Port Server
* SEL-5010 Relay Assistant Software

The [SEL-2032 datasheet](https://www.selinc.com/WorkArea/DownloadAsset.aspx?id=1181) and [SEL-2701 flyer](https://www.selinc.com/WorkArea/DownloadAsset.aspx?id=2918) have more information about the telnet capabilities of these devices.

adepasquale, 2016-02-04:
@infracritical do you have more information about what [these devices on Shodan](https://www.shodan.io/search?query=SEL+telnet+server) might be?