mushorg/conpot #285: Create a general telnet server with templates and custom dialog flow

adepasquale, 2015-11-09:
As suggested by @glaslos it would be cool to have a single telnet server implementation with templates and an additional dialog flow mechanism that would easily allow to add more devices.

Ideally it could be a state diagram where:
* states represent messages sent by the server; and
* (different) transitions are triggered by (different) client input.

adepasquale, 2016-02-02:
@glaslos I think that [telnetsrvlib](https://github.com/ianepperson/telnetsrvlib) could do the job.

It supports gevent StreamServer so it should be easy to fit into Conpot.

glaslos, 2016-02-04:
Ah, this looks great. Should be trivial to use it.

adepasquale, 2016-02-04:
Yes, should be. What if I start working on #282 using `telnersrvlib` and then eventually rewrite the Kamstrup management module with that same library?

xandfury, 2018-01-01:
Team, as per discussion with @glaslos , I think it might be a good idea to start with a simple banner with a login prompt, catching username/pass. 
I am not sure about the dialog/flow. But my suggestion is to display all the connected slaves, network info (displaying Conpot's IP, Gateway, etc. MAC address etc.) etc. and displaying it to the user. 

Some dorking keywords for telnet enabled PLCs to help with this: https://www.slideshare.net/qqlan/internet-connected-icsscadaplc/23