mushorg/conpot #288: Modbus Error: Exception code = 2

cudeso, 2015-12-02:
Hi,

When I run Conpot with a template to only include Modbus (basically the 'default' without snmp, http, ...) I get these errors when using mbtget.
`
mbtget -r1 -u 1 127.0.0.1 -d
`
results in the conpot logs in 
`
Exception caught: Modbus Error: Exception code = 2. (A proper response will be sent to the peer)
`

Unit 1 is defined in the XML file. For info, r1 is for requesting coils, getting the same error when reading holding registers with r3. I also get similar errors when using Modbus Poll (Windows) towards Conpot.

Note:  when I write a test slave directly with modbus-tk I'm getting the same error
```
        server = modbus_tcp.TcpServer(502, "127.0.0.1")
...
        server.start()
        slave_1 = server.add_slave(1)
        slave_1.add_block('0', cst.COILS, 1, 10)
```
Any help on how to fix the responses to the Modbus queries highly appreciated.

koen

glaslos, 2015-12-02:
Can you find out what slave id you are using?

cudeso, 2015-12-03:
Tried with different slave id's (1, 2, 5).

This is the modbus.xml file
```
<modbus enabled="True" host="0.0.0.0" port="502">
    <device_info>
        <VendorName>Siemens</VendorName>
        <ProductCode>SIMATIC</ProductCode>
        <MajorMinorRevision>S7-200</MajorMinorRevision>
    </device_info>
    <mode>serial</mode>
    <delay>100</delay>
    <slaves>
        <slave id="2">
            <blocks>
                <block name="sl2_A">
                    <type>COILS</type>
                    <starting_address>1</starting_address>
                    <size>128</size>
                    <content>sl2_A</content>
                </block>
                <block name="sl2_B">
                    <type>HOLDING_REGISTERS</type>
                    <starting_address>40001</starting_address>
                    <size>32</size>
                    <content>sl2_B</content>
                </block>
            </blocks>
        </slave>
    </slaves>
</modbus>
```

This is the console output from Conpot (removed non relevant) lines. Conpot detects the proper slave id and in the response it sends back the requested function (first request was F1 -> 8102 ; second request was F3 -> 8302)

```
2015-12-03 10:16:21,002 New Modbus connection from 192.168.171.139:46192. (fb86804c-06aa-497c-a53b-e793c78fd350)
ERROR:conpot.protocols.modbus.slave:Exception caught: Modbus Error: Exception code = 2. (A proper response will be sent to the peer)
2015-12-03 10:16:21,002 Exception caught: Modbus Error: Exception code = 2. (A proper response will be sent to the peer)
2015-12-03 10:16:21,003 Modbus traffic from 192.168.171.139: {'function_code': 1, 'slave_id': 2, 'request': 'eaad00000006020100000001', 'response': '8102'} (fb86804c-06aa-497c-a53b-e793c78fd350)

2015-12-03 10:16:32,796 New Modbus connection from 192.168.171.139:46193. (fb86804c-06aa-497c-a53b-e793c78fd350)
ERROR:conpot.protocols.modbus.slave:Exception caught: Modbus Error: Exception code = 2. (A proper response will be sent to the peer)
2015-12-03 10:16:32,797 Exception caught: Modbus Error: Exception code = 2. (A proper response will be sent to the peer)
2015-12-03 10:16:32,797 Modbus traffic from 192.168.171.139: {'function_code': 3, 'slave_id': 2, 'request': 'ad0a00000006020300000001', 'response': '8302'} (fb86804c-06aa-497c-a53b-e793c78fd350)
```

(192.168.171.139 is the IP of the VM, MBTGET is run from the same VM where Conpot runs)

cudeso, 2015-12-17:
After some digging I found this to be more of an issue with the underlying modbus-tk. I implemented a client with pymodbus with similar settings (different unit, coils, hr, ir, etc). The client in pymodbus replied properly whereas with modbus-tk it always failed. 
Requests were done with plcscan, mbtget and the NSE scripts from nmap.

zhiyinyouzhao, 2016-10-19:
Is this bug updae?

glaslos, 2016-10-19:
No update on this so far.

windflyer, 2017-06-21:
This is an issue of reference number. modbus_tk indexes coils with numbers from 1. However, some scripts will read coils from starting address 0. The underlying modbus_tk cannot find coil 0 and thus returns exception code 2: ILLEGAL DATA ACCESS.

creolis, 2017-06-28:
We should run a quick test if modbus-tk accepts starting-address 0,
but if this also works, this issue is closed

windflyer, 2017-06-28:
@creolis Yes, it does. I have confirmed.

The key is how the template file looks like.

creolis, 2017-06-28:
Perfect!

creolis, 2017-06-28:
Thanks Yuru!

closed 2017-06-28 by creolis

fuchengushenxing, 2017-09-05:
Hi，how can I solve this problem? Thanks！

windflyer, 2017-09-05:
@fuchengushenxing Please take a look at my comment posted on Jun 21.

Let me know if you still have questions.

fuchengushenxing, 2017-09-05:
Thanks , i am going to try.