mushorg/conpot #289: Conpot does not respond to SNMPV2c request

chintanhshah, 2015-12-04:
Hi, 

Just checking out there if anybody gets the chance to look at the Conpot SNMP issue . Looks like snmp script responds to the V1 request but does not respond to the v2c request . 

Here is the Conpot's output : 

2015-12-04 14:11:32,122 SNMPv2 Get request from ('10.254.0.1', 60096): 1.3.6.1.2.1.1.6.0
2015-12-04 14:11:49,898 SNMPv2 Get request from ('10.254.0.1', 37841): 1.3.6.1.2.1.1.1.0
2015-12-04 14:40:36,660 SNMPv2 Get request from ('10.254.0.1', 34740): 1.3.6.1.2.1.1.1.0
2015-12-04 15:40:36,178 SNMPv2 Get request from ('10.254.0.1', 46492): 1.3.6.1.2.1.1.1
2015-12-04 15:44:34,278 SNMPv2 Get request from ('10.254.0.1', 35383): 1.3.6.1.2.1.1.1.0
2015-12-04 15:44:57,685 SNMPv2 Get request from ('10.254.0.1', 41628): 1.3.6.1.2.1.1.1.0
2015-12-04 15:48:06,118 SNMPv2 Get request from ('10.254.0.1', 40677): 1.3.6.1.2.1.1.1.0
2015-12-04 15:49:42,890 SNMPv2 Get request from ('10.254.0.1', 42523): 1.3.6.1.2.1.1.1.0
2015-12-04 15:49:51,443 SNMPv2 Get request from ('10.254.0.1', 50911): 1.3.6.1.2.1.1.1
2015-12-04 15:49:54,740 SNMPv2 Get request from ('10.254.0.1', 42575): 1.3.6.1.2.1.1.1

Let me know if there is any way to verify v2c and v3.

glaslos, 2015-12-04:
Are you running the latest version of Conpot? I don't think we made any
major changes but just to be save.

chintanhshah, 2015-12-04:
I am running 0.4.0 ..I was trying few v2c commands and it gives the following error : Some access issue..

[root@Default-CentOS-NBNS ~]# snmpget -v 2c -c public 10.16.3.196 .1.3.6.1.2.1.1.3
Error in packet
Reason: noAccess
Failed object: SNMPv2-MIB::sysUpTime

On the conpot:
2015-12-04 19:56:20,985 SNMPv2 Get request from ('10.16.3.195', 34871): 1.3.6.1.2.1.1.3

PR #314 (2016-12-08): Fix v2c support
Fixes #289 and #292

Before:

```
$  snmpwalk -v2c -c public localhost 
Error in packet.
Reason: authorizationError (access denied to that object)
Failed object: SNMPv2-SMI::mib-2

SNMPv2-SMI::mib-2 = No Such Object available on this agent at this OID
```

After:

```
$  snmpwalk -v2c -c public localhost 
SNMPv2-MIB::sysDescr.0 = STRING: Siemens, SIMATIC, S7-200
SNMPv2-MIB::sysObjectID.0 = OID: SNMPv2-SMI::enterprises.20408
DISMAN-EVENT-MIB::sysUpTimeInstance = Timeticks: (1) 0:00:00.01
SNMPv2-MIB::sysContact.0 = STRING: Siemens AG
SNMPv2-MIB::sysName.0 = STRING: CP 443-1 EX40
 [... cut]
commit d7f9f1ec (2016-12-08): Fix v2c support (#314)

Fixes #289 and #292
closed 2016-12-08 by glaslos