mushorg/conpot #290: Investigate databus issues on default S7-200 template

adepasquale, 2015-12-11:
http://xiphosresearch.com/2015/12/09/OPSEC-For-Honeypots.html

https://github.com/mushorg/conpot/blob/33363ab658846ad72ffac2525ed449152b9c4b1f/conpot/templates/default/template.xml#L40-L60
https://github.com/mushorg/conpot/blob/33363ab658846ad72ffac2525ed449152b9c4b1f/conpot/core/databus.py#L85-L86

glaslos, 2015-12-11:
Is this actually an issue from our side or an issue from the people setting up Conpot?

adepasquale, 2015-12-11:
Definitely a setup issue (e.g. people using default serial numbers).

Anyway I'd like to double check to make sure there are NO databus issues.

For instance, the one below:
```
PLC name: PG[random.randint(0,1) for  # ...
```

glaslos, 2015-12-11:
Isn't that https://github.com/mushorg/conpot/blob/33363ab658846ad72ffac2525ed449152b9c4b1f/conpot/templates/default/template.xml#L32 which should be a string?

adepasquale, 2015-12-11:
Yes, right. The [SystemName value](https://github.com/mushorg/conpot/blob/33363ab658846ad72ffac2525ed449152b9c4b1f/conpot/templates/default/template.xml#L16-L18) is probably misconfigured.

closed 2015-12-11 by adepasquale

creolis, 2015-12-11:
though this is closed now, just a comment regarding the String on https://github.com/mushorg/conpot/blob/33363ab658846ad72ffac2525ed449152b9c4b1f/conpot/templates/default/template.xml#L32 - it *might* make sense to introduce some sort of variable parts to such lines, like the guy that misconfigured conpot might have assumed.

Maybe even a generator that produces a semi-random (random per installation) string that stays the same on the system for things like serial numbers. That way, we would have different serial numbers on each fresh setup that won't change on restarting conpot and won't change on every other request.

Second, I think we should consider a post-setup dialog / first-start dialog that screams "change these default values" into their face and maybe even offer some sort of wizzard (yeah, with 2 z's) in order to make it easier to get away from defaults.

Anything that lowers the number of deployed mouser factory conpots ;-)

glaslos, 2015-12-13:
I don't think that randomizing everything by default is a good way to go. This would force us to do complex randomization and those searching for honeypots to write some easy regex. I like your approach of educating the user and refusing to start Conpot until you have changed all the default settings.

johnnykv, 2015-12-13:
Deceptions requires thought process - it's a art that requires practice and skill. One of the reasons I put in those obviously wrong default values was to force researchers to customise it, if anyone (hello security vendors) decides to deploy Conpot with default values they deserve to have their honeypots burned.

adepasquale, 2015-12-15:
I am all for refusing to execute if people don't take time to configure they honeypots. Otherwise we end up with stuff like this (from http://news.softpedia.com/news/researchers-fail-at-setting-up-proper-honeypots-497574.shtml) 

> Mr. Martyn identified the root of the problem as being ConPot, an open source toolkit that allows infosec researchers to quickly install an ICS/SCADA honeypot on their servers.
>Apparently, the tool comes with hardcoded identification details, that most security researchers forget to change.

glaslos, 2015-12-15:
".. hardcoded identification details, that most security researchers forget to change." (facepalm)

creolis, 2015-12-15:
yeah .. those XML templates were really hard to code ... *coughs*