mushorg/conpot #291: Stop startup if default configuration

glaslos, 2015-12-13:
As discussed in #290 we want to stop startup if the default configuration has not been modified.

adepasquale, 2015-12-15:
Should we check that values are different from the default or, more easily, clear the default template and check for empty values there?

glaslos, 2015-12-15:
I'd just add a Boolean in the template that needs to be set to True to make Conpot run.

glaslos, 2015-12-15:
Add a comment to that Boolean in all capitals that you should change your default template.

johnnykv, 2015-12-15:
1. Stop Conpot if the hash of the current config file equals the hash of the distributed config file. This does not ensure the all options has been changed, but it is really our job stopping stupid people from being stupid?
2. Implement '-f' option to disable the check above.

adepasquale, 2015-12-15:
@johnnykv I like option 1 but not so sure about 2.

johnnykv, 2015-12-15:
@adepasquale When developing we need to be able to run with default config.

glaslos, 2015-12-15:
I'm pro the -f, also helps with tests

adepasquale, 2015-12-15:
How about running with a dedicated testing configuration?

glaslos, 2017-09-10:
How about renaming the default configuration to testing and use the -f to force conpot to run on a non-localhost interface with the testing config?

xandfury, 2018-01-07:
I would also suggest a note on the README that ***screams*** to the user not to run the default config on production servers. Something like:

#### NOTE: DO NOT RUN DEFAULT CONFIG ON PRODUCTION SYSTEMS

PR #350 (2018-01-10): Easy Add -f option to use testing.cfg. 
Fix #291
closed 2018-01-10 by creolis