mushorg/conpot #292: SNMP V2c authorization error with Conpot

chintanhshah, 2015-12-24:
Hi Folks , 

I was trying to verify the working of SNMP v2c with conpot but looks like the server script throws "authorization error" for SNMP V2c command issued  . 

[root@Default-CentOS-NBNS ~]# snmpwalk -c public -v 2c 10.254.0.2
Error in packet.
Reason: authorizationError (access denied to that object)
Failed object: SNMPv2-SMI::mib-2
SNMPv2-SMI::mib-2 = No Such Object available on this agent at this OID

I've tried with relevant SNMP V2 MIB (using OID ) as well but it doesn't respond as expected . 

Could somebody take a look at this and let me know how to verify the working of SNMP v2c  ?

glaslos, 2015-12-26:
Hey, are you able to share the output from Conpot for that command?

chintanhshah, 2015-12-27:
Below is the output from the Conpot for the above command : 

2015-12-27 14:07:19,013 SNMPv2 GetNext request from ('10.254.0.1', 56707): 1.3.6.1.2.1
2015-12-27 14:07:19,113 SNMPv2 Get request from ('10.254.0.1', 56707): 1.3.6.1.2.1
2015-12-27 14:07:19,114 SNMPv2 response to ('10.254.0.1', 56707): 1.3.6.1.2.1

Let me know if anybody can throw some light ..

katkad, 2016-01-05:
similar to https://github.com/mushorg/conpot/issues/289

glaslos, 2016-12-08:
Fixed by #314

closed 2016-12-08 by glaslos