mushorg/conpot #294: Connecting Conpot with MATLAB

Tholep, 2016-04-30:
I am not sure whether this is a right place to post a question about Conpot, however, I can't find any Conpot community available.

I am working on project that a simple plan ( e.g water tank plan) is simulated in MATLAB and sensors sends volts to a simulated PLC via Modbus. I find that Conpot is an interesting honeypot, however, I am not sure how to integrate between Conpot and MATLAB. Can anyone help to give hints to achieve the goal?

Thanks.

glaslos, 2016-04-30:
Have you tried pointing matlab to Conpot? Conpot should be able to see the writes.

Tholep, 2016-05-01:
No, I haven't because I have no idea how to do that for now, can you provide some hints how to do that? At the moment, I  have a simple Conpot box simulating Siemens S7-200 PLC and it connects to syslog and plunk servers for report.
Thanks for your help.

glaslos, 2016-05-01:
I have no experience with MATLAB but have you tried to make a TCP connection on port 502 to Conpot and read some data?

Tholep, 2016-05-02:
I don't think I get your question. To my understanding, Conpot simulates a PLC and Modbus protocol, so my idea is to make Conpot as a master and connect it to other slaves ( e.g. sensors) to read data like in ICS plan. Then based on input data, PLC (Conpot) will issue appropriate command to its slaves. Please correct me, if I misunderstand something (This is my first time working with industrial plan). I appreciate you advice  and help.

P/s: at the moment, I am trying to establish connection between Conpot as master and two simulated slaves (http://www.modbustools.com/download.html) via Modbus TCP.

Tholep, 2016-05-03:
I think I understand you now. I also use a Modbus master (http://www.modbustools.com/download.html)  to read data from Conpot. However, in the simulated modbus master, it can connect to Conpot - Slave id 0 ( address 1 function 01). Conpot reports "Modbus response sent to 192.168.1.2", but the simulated master consistently report : "slave device fail".
When I try to connect to conpot-slave ID 1 ( address 1 function code 01), I suppose the result should be similar to slave 0, however, the master always gets error to read COILS, while Conpot reports:

"
2016-05-03 06:47:15,809 New Modbus connection from 192.168.1.2:50715. (7af57ade-c9ae-4e41-818e-e336a852e49d)
2016-05-03 06:47:16,827 Modbus traffic from 192.168.1.2: {'function_code': 1, 'slave_id': 1, 'request': '013e0000000601010001000a', 'response': ''} (7af57ade-c9ae-4e41-818e-e336a852e49d)
2016-05-03 06:47:16,827 Modbus client ignored due to invalid addressing. (7af57ade-c9ae-4e41-818e-e336a852e49d)
"

closed 2016-08-24 by Tholep

glaslos, 2016-08-24:
Did you manage to connect MATLAB to Conpot?

Tholep, 2016-08-24:
we changed our plan in which MATLAB does not talk directly to Conpot. However, there are libraries in Matlab that facilitates Modbus communication (chapter 7 Rocking the pocket book:Hacking chemical plants for competition and extortion by Marina Krotofil), so I believe, it should not be a problem to couple those directly.