mushorg/conpot #298: Missing dependency?

Bifrozt, 2016-07-17:
Hi and thanks for an awesome project!

I tried installing conpot, both dev and pip version, and received the same error when running `python setup.py install` or `pip install conpot`

```
Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "/tmp/pip_build_root/MySQL-python/setup.py", line 17, in <module>

    metadata, options = get_config()

  File "/tmp/pip_build_root/MySQL-python/setup_posix.py", line 43, in get_config

    libs = mysql_config("libs_r")

  File "/tmp/pip_build_root/MySQL-python/setup_posix.py", line 25, in mysql_config

    raise EnvironmentError("%s not found" % (mysql_config.path,))

EnvironmentError: mysql_config not found
```

It's easy enough to fix by running `sudo apt-get install libmysqlclient-dev`.

The question is, did i not read the doc well enough, is this installed by default in Ubuntu 12.04 or has this dependency been missed in the docs?

(If its me not reading the docs well enough, feel free to ignore me and close this ticket.)

---

**OS**:
```
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.4 LTS
Release:	    14.04
Codename:	    trusty
```

glaslos, 2016-07-20:
Interesting, I was expecting `apt-get install python-mysqldb` from the documentation to cover the `pip install MySQL-python` requirement. Could you verify if one of them is redundant?
I guess we should make all the logging options actually optional when it comes to the requirements for installation, similar to #62

Bifrozt, 2016-07-20:
Sure, ill check it tonight and let you know

creolis, 2016-07-20:
Yup, we should not load client libs as long as their underlying feature is not enabled by configuration. This will keep it sleeker in first place.

Bifrozt, 2016-07-23:
Installing dependencies (http://mushorg.github.io/conpot/installation/ubuntu.html)
```
$ sudo apt-get install libsmi2ldbl snmp-mibs-downloader python-dev libevent-dev libxslt1-dev libxml2-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  binutils cpp cpp-4.8 gcc gcc-4.8 libasan0 libatomic1 libc-dev-bin libc6-dev
  libcloog-isl4 libevent-core-2.0-5 libevent-extra-2.0-5
  libevent-openssl-2.0-5 libevent-pthreads-2.0-5 libexpat1-dev libgcc-4.8-dev
  libgomp1 libisl10 libitm1 libmpc3 libmpfr4 libpython-dev libpython2.7-dev
  libquadmath0 libtsan0 libxslt1.1 linux-libc-dev manpages-dev python2.7-dev
  smistrip
Suggested packages:
  binutils-doc cpp-doc gcc-4.8-locales gcc-multilib make autoconf automake1.9
  libtool flex bison gdb gcc-doc gcc-4.8-multilib gcc-4.8-doc libgcc1-dbg
  libgomp1-dbg libitm1-dbg libatomic1-dbg libasan0-dbg libtsan0-dbg
  libquadmath0-dbg glibc-doc pkg-config unzip
The following NEW packages will be installed:
  binutils cpp cpp-4.8 gcc gcc-4.8 libasan0 libatomic1 libc-dev-bin libc6-dev
  libcloog-isl4 libevent-core-2.0-5 libevent-dev libevent-extra-2.0-5
  libevent-openssl-2.0-5 libevent-pthreads-2.0-5 libexpat1-dev libgcc-4.8-dev
  libgomp1 libisl10 libitm1 libmpc3 libmpfr4 libpython-dev libpython2.7-dev
  libquadmath0 libsmi2ldbl libtsan0 libxml2-dev libxslt1-dev libxslt1.1
  linux-libc-dev manpages-dev python-dev python2.7-dev smistrip
  snmp-mibs-downloader
0 upgraded, 36 newly installed, 0 to remove and 0 not upgraded.
Need to get 48.3 MB of archives.
After this operation, 119 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```
Installing conpot
```
udo pip install conpot
Downloading/unpacking conpot
  Downloading Conpot-0.5.1.tar.gz (111kB): 111kB downloaded
  Running setup.py (path:/tmp/pip_build_root/conpot/setup.py) egg_info for package conpot
    
Downloading/unpacking gevent>=1.0 (from conpot)
  Downloading gevent-1.1.2.tar.gz (2.0MB): 2.0MB downloaded
  Running setup.py (path:/tmp/pip_build_root/gevent/setup.py) egg_info for package gevent
    
    no previously-included directories found matching '*/__pycache__'
    warning: no previously-included files matching '*.so' found anywhere in distribution
    warning: no previously-included files matching '*.o' found anywhere in distribution
    warning: no previously-included files matching 'config.log' found anywhere in distribution
    warning: no previously-included files matching 'config.status' found anywhere in distribution
    no previously-included directories found matching 'doc/_build'
    warning: no previously-included files matching '*.pyc' found anywhere in distribution
    warning: no previously-included files matching '.coverage' found under directory 'greentest'
    no previously-included directories found matching 'greentest/htmlcov'
    warning: no previously-included files matching 'stamp-h?' found under directory 'c-ares'
    warning: no previously-included files matching 'ares_build.h.orig' found under directory 'c-ares'
    no previously-included directories found matching 'libev/.deps'
    warning: no previously-included files matching 'Makefile' found under directory 'libev'
    warning: no previously-included files matching 'libtool' found under directory 'libev'
    warning: no previously-included files matching 'stamp-h?' found under directory 'libev'
    warning: no previously-included files matching 'config.h' found under directory 'libev'
    warning: no previously-included files matching '_corecffi.c' found under directory 'gevent'
    warning: no previously-included files found matching 'Makefile'
    
    WARNING: The gevent.ares extension has been disabled.
Downloading/unpacking pysnmp==4.2.5 (from conpot)
  Downloading pysnmp-4.2.5.tar.gz (225kB): 225kB downloaded
  Running setup.py (path:/tmp/pip_build_root/pysnmp/setup.py) egg_info for package pysnmp
    
Downloading/unpacking lxml (from conpot)
  Downloading lxml-3.6.0.tar.gz (3.7MB): 3.7MB downloaded
  Running setup.py (path:/tmp/pip_build_root/lxml/setup.py) egg_info for package lxml
    Building lxml version 3.6.0.
[... 86 lines cut ...]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 33: ordinal not in range(128)
```

`pip install MySQL-python` is redundant after installing `python-mysql`
 
<hr>

Attempting to resolve issue by installing `libmysqlclient-dev`

```
sudo apt-get install libmysqlclient-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  zlib1g-dev
The following NEW packages will be installed:
  libmysqlclient-dev zlib1g-dev
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 1,048 kB of archives.
After this operation, 6,059 kB of additional disk space will be used.
Do you want to continue? [Y/n]
```

`sudo pip install conpot` can be installed and runs as expected.

glaslos, 2016-07-23:
I removed the `pip install MySQL-python` from the `requirements.txt`. `apt install python-mysql libmysqlclient-dev` is already in the Ubuntu install instructions.
Thanks for the feedback, I'm closing the issue assuming this clarified.

closed 2016-07-23 by glaslos