mushorg/conpot #299: ImportError: No module named MySQLdb

wangyufeng0615, 2016-07-25:
I've deployed conpot by MHN but conpot can't start running.

supervisorctl status
`conpot    FATAL    Exited too quickly (process log may have details)`

I checked /var/log/conpot.out:

```
Traceback (most recent call last):
  File "/opt/conpot/env/bin/conpot", line 6, in <module>
    exec(compile(open(__file__).read(), __file__, 'exec'))
  File "/opt/conpot/env/src/conpot/bin/conpot", line 39, in <module>
    from conpot.core.loggers.log_worker import LogWorker
  File "/opt/conpot/env/local/lib/python2.7/site-packages/gevent/builtins.py", line 93, in __import__
    result = _import(*args, **kwargs)
  File "/opt/conpot/env/src/conpot/conpot/core/loggers/log_worker.py", line 30, in <module>
    from conpot.core.loggers.mysql_log import MySQLlogger
  File "/opt/conpot/env/local/lib/python2.7/site-packages/gevent/builtins.py", line 93, in __import__
    result = _import(*args, **kwargs)
  File "/opt/conpot/env/src/conpot/conpot/core/loggers/mysql_log.py", line 19, in <module>
    import MySQLdb
  File "/opt/conpot/env/local/lib/python2.7/site-packages/gevent/builtins.py", line 93, in __import__
    result = _import(*args, **kwargs)
ImportError: No module named MySQLdb
```

I tried `apt-get install python-mysqldb` and `pip install mysql-python` but I already intalled this.
Also I tried install mysql-python by source code. The installation is success but conpot can't running with same reason.

Thanks for your help!

wangyufeng0615, 2016-07-25:
And, if I type 'import MySQLdb' in normal python terminal, it's ok.

glaslos, 2016-07-25:
Can you pull the latest master?

wangyufeng0615, 2016-07-25:
Thank you. It seems like I re-run the deploy script and re-download the conpot, everything is ok now.

closed 2016-07-25 by glaslos