mushorg/conpot #300: compilation error on raspberry pi raspbian Jessie

gento, 2016-08-02:
Hi,

I tried to compile conpot on raspberry pi rasbian Jessie (release 2016-05-27) and the compilation looks successful. However, when I try to run conpot, it always hit me with this error message:

pi@raspberrypi:/conpot $ sudo conpot
Traceback (most recent call last):
  File "/usr/local/bin/conpot", line 39, in <module>
    from conpot.core.loggers.log_worker import LogWorker
  File "/usr/local/lib/python2.7/dist-packages/conpot/core/loggers/log_worker.py", line 33, in <module>
    from conpot.core.loggers.taxii_log import TaxiiLogger
  File "/usr/local/lib/python2.7/dist-packages/conpot/core/loggers/taxii_log.py", line 25, in <module>
    from conpot.core.loggers.stix_transform import StixTransformer
  File "/usr/local/lib/python2.7/dist-packages/conpot/core/loggers/stix_transform.py", line 25, in <module>
    from stix.core import STIXHeader, STIXPackage
  File "build/bdist.linux-armv7l/egg/stix/__init__.py", line 5, in <module>
  File "build/bdist.linux-armv7l/egg/stix/base.py", line 15, in <module>
  File "build/bdist.linux-armv7l/egg/stix/utils/__init__.py", line 379, in <module>
  File "build/bdist.linux-armv7l/egg/stix/utils/nsparser.py", line 434, in <module>
TypeError: 'NamespaceSet' object is not iterable

pi@raspberrypi:/honeeepi $ python
Python 2.7.9 (default, Mar  8 2015, 00:52:26) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> 

Have anyone seen this before? Any idea how I can resolve this? 
Thanks!

Cheer,
Tan

glaslos, 2016-08-02:
We haven't pinned the version of stix in the requests.txt, could you check if there was a change in the library?

glaslos, 2016-08-02:
Also check the installed version of stix using `pip freeze | grep stix`
Maybe a `pip2 install -U requirements.txt` is all you need.

gento, 2016-08-02:
This is the installed version of stix that I had in Raspbian Jessie:

pi@raspberrypi:/conpot $ pip freeze | grep stix
stix==1.2.0.1.dev1
stix-validator==2.4.0

pi@raspberrypi:/conpot $ pip show stix
---
Name: stix
Version: 1.2.0.1.dev1
Location: /usr/local/lib/python2.7/dist-packages/stix-1.2.0.1.dev1-py2.7.egg
Requires: lxml, python-dateutil, cybox, mixbox
pi@raspberrypi:/conpot $ pip show stix-validator
---
Name: stix-validator
Version: 2.4.0
Location: /usr/local/lib/python2.7/dist-packages/stix_validator-2.4.0-py2.7.egg
Requires: lxml, xlrd, ordereddict, python-dateutil
pi@raspberrypi:/conpot $

glaslos, 2016-08-02:
Can you install the 1.2.0.0?

gento, 2016-08-02:
Thanks Lukas! You are the great as usual!

I downgraded stix and cybox to lower version and it works as charming as ever

- stix==stix==1.2.0.1.dev1 to stix=1.2.0.0
- cybox==2.1.0.13.dev1 to cybox==2.1.0.12

```
pi@raspberrypi:/conpot $ sudo pip install 'stix>=1.1.1.5,<1.2.0.1'
<forgot to take note about the output...>

pi@raspberrypi:/honeeepi $ sudo conpot
Traceback (most recent call last):
  File "/usr/local/bin/conpot", line 39, in <module>
    from conpot.core.loggers.log_worker import LogWorker
  File "/usr/local/lib/python2.7/dist-packages/conpot/core/loggers/log_worker.py", line 33, in <module>
    from conpot.core.loggers.taxii_log import TaxiiLogger
  File "/usr/local/lib/python2.7/dist-packages/conpot/core/loggers/taxii_log.py", line 25, in <module>
    from conpot.core.loggers.stix_transform import StixTransformer
  File "/usr/local/lib/python2.7/dist-packages/conpot/core/loggers/stix_transform.py", line 25, in <module>
    from stix.core import STIXHeader, STIXPackage
  File "/usr/local/lib/python2.7/dist-packages/stix/__init__.py", line 5, in <module>
    from .base import (Entity, EntityList, TypedCollection, TypedList,  # noqa
  File "/usr/local/lib/python2.7/dist-packages/stix/base.py", line 11, in <module>
    from . import bindings, utils
  File "/usr/local/lib/python2.7/dist-packages/stix/utils/__init__.py", line 388, in <module>
    from .nsparser import *  # noqa
  File "/usr/local/lib/python2.7/dist-packages/stix/utils/nsparser.py", line 434, in <module>
    (ns, loc) for ns, _, loc in cybox.utils.nsparser.NS_LIST if loc
AttributeError: 'module' object has no attribute 'NS_LIST'
pi@raspberrypi:/conpot $ pip freeze | grep cybox
cybox==2.1.0.13.dev1
pi@raspberrypi:/conpot $ sudo pip install "cybox==2.1.0.12"
Downloading/unpacking cybox==2.1.0.12
  Downloading cybox-2.1.0.12.zip (792kB): 792kB downloaded
  Running setup.py (path:/tmp/pip-build-dPiH1y/cybox/setup.py) egg_info for package cybox
    
    warning: no previously-included files matching '*.pyc' found under directory 'docs'
    warning: no previously-included files matching '*.pyo' found under directory 'docs'
    no previously-included directories found matching 'docs/_build'
    warning: no previously-included files matching '*.pyc' found under directory 'examples'
    warning: no previously-included files matching '*.pyo' found under directory 'examples'
Requirement already satisfied (use --upgrade to upgrade): lxml>=2.2.3 in /usr/lib/python2.7/dist-packages (from cybox==2.1.0.12)
Requirement already satisfied (use --upgrade to upgrade): python-dateutil in /usr/lib/python2.7/dist-packages (from cybox==2.1.0.12)
Requirement already satisfied (use --upgrade to upgrade): six in /usr/local/lib/python2.7/dist-packages/six-1.10.0-py2.7.egg (from python-dateutil->cybox==2.1.0.12)
Installing collected packages: cybox
  Found existing installation: cybox 2.1.0.13.dev1
    Uninstalling cybox:
  Running setup.py install for cybox
    
    warning: no previously-included files matching '*.pyc' found under directory 'docs'
    warning: no previously-included files matching '*.pyo' found under directory 'docs'
    no previously-included directories found matching 'docs/_build'
    warning: no previously-included files matching '*.pyc' found under directory 'examples'
    warning: no previously-included files matching '*.pyo' found under directory 'examples'
  Could not find .egg-info directory in install record for cybox==2.1.0.12
Successfully installed cybox
Cleaning up...
pi@raspberrypi:/honeeepi $ sudo conpot

                       _
   ___ ___ ___ ___ ___| |_
  |  _| . |   | . | . |  _|
  |___|___|_|_|  _|___|_|
              |_|

  Version 0.5.1
  MushMush Foundation

...
```

glaslos, 2016-08-02:
If you have too much time, send a pull request with fixed versions in the requirements.txt for stix and cybox 😄

closed 2016-08-02 by glaslos