mushorg/conpot #302: Conpot with  mysql log

Tholep, 2016-08-24:
Hi,

While i am working with conpot and mysql for logging, i figure out two issues with mysql_log.py and would like to report as below:

1. _connect() method calls  _create_db() everytime an instance of this class created. However the  _create_db() create table "events" without checking whether that table exists or not. This causes conpot hanging because log_worker greenlet dies. Below is my fix:

 # check whether table 'events' exists
        cursor.execute(""" SELECT count(*) FROM information_schema.tables WHERE table_name = 'events'""")
        if self.conn.fetchone()[0]==0:
            cursor.execute("""CREATE TABLE IF NOT EXISTS `events` (
                            `id` bigint(20) NOT NULL AUTO_INCREMENT,
                            `sensorid` text NOT NULL,
                            `session` text NOT NULL,
                            `timestamp` timestamp NOT NULL DEFAULT CURRENT_TIMESTAMP,
                            `remote` text NOT NULL,
                            `protocol` text NOT NULL,
                            `request` text NOT NULL,
                            `response` text NOT NULL,
                            PRIMARY KEY (`id`)
                            ) ENGINE=InnoDB DEFAULT CHARSET=latin1;
                           """)

2. Whenever a Modbus request arrives Conpot, there will be 3 events: "connection_new", "modbus function code" and "connection_lost". Therefore, event["data"] does not always have keys : request and response, which will trigger an exception ('Logging failed. Database connection not available.'). below is my fix for Modbus

if "function_code" in event["data"].keys():
    cursor.execute("""INSERT INTO
                        events (sensorid, session, remote, protocol, request, response)
                      VALUES
                        (%s, %s, %s, %s,%s, %s)""", (str(self.sensorid),
                                                      str(event["id"]),
                                                      str(event["remote"]),
                                                      event["data_type"],
                                                      event["data"].get('request'),
                                                      event["data"].get('response')))
else:
    cursor.execute("""INSERT INTO
                        events (sensorid, session, remote, protocol, request, response)
                      VALUES
                        (%s, %s, %s, %s,%s,"NA")""", (str(self.sensorid),
                                                      str(event["id"]),
                                                      str(event["remote"]),
                                                      event["data_type"],
                                                      event["data"].get('type')))

glaslos, 2016-08-24:
Thanks for reporting those issues. Would you mind sending those fixes as pull requests?

Tholep, 2016-08-24:
sure, I will do it

creolis, 2017-06-28:
Fix merged Dec. 2016

closed 2017-06-28 by creolis