mushorg/conpot #305: Modbus Exception Code:Slave device failure(4)

zhiyinyouzhao, 2016-10-12:
Hi,

When I run Conpot with a default template.
I connetct the server runs Conpot,the console output.

2016-10-12 13:34:08,216 Modbus response sent to 192.168.2.2
2016-10-12 13:34:09,229 Modbus traffic from 192.168.2.2: {'function_code': None, 'slave_id': 1, 'request': '990d0000000601010000000a', 'response': ''} (dd1115e2-5135-4287-8761-9b1499d629b6)

I think the 'function_code:None'is error,but I don't know the reason.

Then I use  Wireshark to catch the data package.
The information:
Function 1:Read Coils.Exception:Slave device failure
	Function Code:Unknown(129)
	Exception Code:Slave device failure(4)

Note:VM runs conpot IP:192.168.2.3
Any help for me to fix the probles of Modbus.thanks a lot.

zhiyinyouzhao, 2016-10-13:
Use ModScan2 client to connect conpot.
The modbus client infomation:
Address:0001 
Decice ID:1
Modbus Point Type:01 COIL STAUS

I analysis the slave_db.py.when the slave_id is 1-247,the  line code "slave = self.get_slave(slave_id)" can't execute. So,the procedure access to the except.
Last,the procedure access to if slave: .....
and the return function_code=none.So,it seems correctly.
Please help to confirm whether it is true.

slave_db.py about lines code:
                    elif slave_id > 0 and slave_id <= 247:
                    slave = self.get_slave(slave_id)
                    response_pdu = slave.handle_request(request_pdu)
                    # make the full response
                    response = query.build_response(response_pdu)
                else:
                    # return no response, and data necessary for logging
                    return (None, {'request': request_pdu.encode('hex'),
                                   'slave_id': slave_id,
                                   'function_code': func_code,
                                   'response': ''})
        except (IOError, MissingKeyError) as e:
            # If the request was not handled correctly, return a server error
            # response
            #
            r = struct.pack(
                ">BB", func_code + 0x80, defines.SLAVE_DEVICE_FAILURE)
            response = query.build_response(r)
      except (ModbusInvalidRequestError) as e:
            logger.info(e)

        if slave:
            function_code = slave.function_code
        return (response, {'request': request_pdu.encode('hex'),
                           'slave_id': slave_id,
                           'function_code': function_code,
                           'response': response_pdu.encode('hex')})

windflyer, 2017-06-03:
Conpot Modbus server by default works in serial mode. Current implementation doesn't handle missing UID properly.
When there is no Modbus device behind a certain UID, we should just return no response instead of a device failure response.

closed 2017-06-28 by creolis