mushorg/conpot #306: Error：add_slave() takes exactly 2 arguments (4 given)

zhiyinyouzhao, 2016-10-14:
Hi，
When I run conpot --template default，I found the error inforamtion in console line.
console output:
2016-10-14 15:27:09,722 Starting Conpot using template: /home/zn/test_env/lib/python2.7/site-packages/conpot/templates/default
2016-10-14 15:27:09,722 Starting Conpot using configuration found in: /home/zn/test_env/lib/python2.7/site-packages/conpot/conpot.cfg
......
2016-10-14 15:27:18,926 add_slave() takes exactly 2 arguments (4 given)
......
add_slave() takes exactly 2 arguments (4 given) 
I test the code，I found  line 69,'slave = self.add_slave(slave_id)'  in modbus_server.py
It can't execute,so the progress logger.info(exception)
Please help me confirm this issues is True or False，thanks.

glaslos, 2016-10-17:
Can you run `pip freeze | grep modbus` and paste the output here?

zhiyinyouzhao, 2016-10-18:
sudo pip freeze | grep modbus:
modbus-tk==0.5.0

glaslos, 2016-10-18:
Uninstall modbus-tk and install it from here: https://github.com/mushorg/modbus-tk

zhiyinyouzhao, 2016-10-19:
It's ok! thanks a lot.

closed 2016-10-19 by glaslos