mushorg/conpot #308: Conpot tests break with bacpypes 0.14

glaslos, 2016-11-08:
```Python
test_whoHas (conpot.tests.test_bacnet_server.TestBase) ... Traceback (most recent call last):
  File "/home/travis/virtualenv/python2.7.9/lib/python2.7/site-packages/gevent-1.2a1-py2.7-linux-x86_64.egg/gevent/greenlet.py", line 536, in run
    result = self._run(*self.args, **self.kwargs)
  File "/home/travis/virtualenv/python2.7.9/lib/python2.7/site-packages/gevent-1.2a1-py2.7-linux-x86_64.egg/gevent/baseserver.py", line 26, in _handle_and_close_when_done
    return handle(*args_tuple)
  File "/home/travis/build/mushorg/conpot/conpot/protocols/bacnet/bacnet_server.py", line 83, in handle
    self.bacnet_app.indication(apdu, address, self.thisDevice)
  File "/home/travis/build/mushorg/conpot/conpot/protocols/bacnet/bacnet_app.py", line 262, in indication
    request.decode(apdu)
  File "/home/travis/virtualenv/python2.7.9/lib/python2.7/site-packages/bacpypes-0.14.1-py2.7.egg/bacpypes/apdu.py", line 689, in decode
    Sequence.decode(self, self._tag_list)
  File "/home/travis/virtualenv/python2.7.9/lib/python2.7/site-packages/bacpypes-0.14.1-py2.7.egg/bacpypes/constructeddata.py", line 266, in decode
    value.decode(taglist)
  File "/home/travis/virtualenv/python2.7.9/lib/python2.7/site-packages/bacpypes-0.14.1-py2.7.egg/bacpypes/constructeddata.py", line 197, in decode
    raise InvalidTag("%s expected context tag %d" % (element.name, element.context))
InvalidTag: deviceInstanceRangeLowLimit expected context tag 0
```

willemtoledano, 2017-01-18:
Also with 0.15 - Rolling back to 0.13.6 worked for me so far

windflyer, 2017-08-01:
This should have been fixed. We already upgraded  to bacpypes 0.16.1

glaslos, 2017-08-01:
You are right!

closed 2017-08-01 by glaslos