mushorg/conpot #310: Cannot access device information when conpot being scanned by Plcscan or nmap(using modicon-info)

Joris-Fu, 2016-11-15:
When testing the authenticity of conpot version 0.5.1 by using PLcscan or Digital Bond Redpoint's modicon-info.nse. We look forward to getting the response like that:
**-------with plcscan, we hope that :**  
```
python plcscan.py --brute-uid 166.130.115.39
Scan start... 
166.130.115.39:502 Modbus/TCP          
  Unit ID: 0
    Device: Schneider Electric   BMX P34 2020 v2.5 
  Unit ID: 255
    Device: Schneider Electric   BMX P34 2020 v2.5 
  Unit ID: 1
    Device: Schneider Electric   BMX P34 2020 v2.5 
  Unit ID: 2
    Device: Schneider Electric   BMX P34 2020 v2.5 
```
**-------But actually, the result is disappointing:**
```
192.168.2.151:502 Modbus/TCP           
  Unit ID: 1
    Device info error: SLAVE DEVICE FAILURE
  Unit ID: 2
    Device info error: SLAVE DEVICE FAILURE
  Unit ID: 3
    Device info error: SLAVE DEVICE FAILURE
  Unit ID: 4
    Device info error: SLAVE DEVICE FAILURE
  Unit ID: 5
    Device info error: SLAVE DEVICE FAILURE
  Unit ID: 6
    Device info error: SLAVE DEVICE FAILURE
```
**I can only get the response from Unit ID 1-247, Why the Unit 0 and Unit 255 are inaccessable, and I cannot figure out why conpot send back SLAVE DEVICE FAILURE**
**-------With modicon-info.nse, we hope to get result like that:**
```
Starting Nmap 7.31 ( https://nmap.org ) at 2016-11-15 23:23 CST
Nmap scan report for mobile-166-130-115-39.mycingular.net (166.130.115.39)
Host is up (0.36s latency).
PORT    STATE SERVICE VERSION
502/tcp open  Modbus
| modicon-info: 
|   Vendor Name: Schneider Electric  
|   Network Module: BMX P34 2020
|   CPU Module: BMX P34 2020
|   Firmware: v2.5
|   Memory Card: BMXRMS008MP
|   Project Information: Station - V6.0   PATRIOTTECH-PC C:\Users\chris.ault\Dropbox (Patriot)\Patriot Automation\patriotmasterfiles\Nabors\Nabors University 31-04\Unload\PLC\university 34-1.stu
|   Project Revision: 0.0.114
|_  Project Last Modified: 9/2/2016 11:11:16
```
**-------But actually, we got nothing:**
```
Starting Nmap 7.31 ( https://nmap.org ) at 2016-11-15 23:29 CST
Nmap scan report for 192.168.2.151
Host is up (0.00015s latency).
PORT    STATE SERVICE VERSION
502/tcp open  mbap?
MAC Address: 08:00:27:DD:C9:3C (Oracle VirtualBox virtual NIC)
```
At conpot side, we found some strange error when being scanning by modicon-info.nse：
```
2016-11-15 23:29:33,847 New Modbus connection from 192.168.2.148:42844. (0e64cca4-977b-4111-93a6-5e6032d34fbc)
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/gevent/greenlet.py", line 534, in run
    result = self._run(*self.args, **self.kwargs)
  File "/usr/local/lib/python2.7/dist-packages/gevent/baseserver.py", line 25, in _handle_and_close_when_done
    return handle(*args_tuple)
  File "/usr/local/lib/python2.7/dist-packages/conpot/protocols/modbus/modbus_server.py", line 104, in handle
    tr_id, pr_id, length = struct.unpack(">HHH", request[:6])
error: unpack requires a string argument of length 6
<Greenlet at 0x7f6055b5ce10: _handle_and_close_when_done(<bound method ModbusServer.handle of <conpot.proto, <bound method StreamServer.do_close of <StreamServ, (<socket at 0x7f6055ab38d0 fileno=[Errno 9] Bad fi)> failed with error

```

windflyer, 2017-06-23:
The root cause is that the client side closes socket too quickly.

While the server (Conpot side) is still processing the previous incoming data, the client side has already sent out all data and closes the socket. The server sides continues reads data from the socket which has been closed and this exception happens.

creolis, 2017-06-28:
corrected via exception handling. I'm closing this, together with it's "sister" issue #169

closed 2017-06-28 by creolis