mushorg/conpot #311: What's the meaning of "No Proxy template found.Service will remain unconfigured/stopped"

Kilento, 2016-12-01:
Hey!
   I am a novice to use conpot.I'm sure it's a good honeypot in the area of industrial control system.However it's hard to search some useful documents about using conpot to help study. It would be very helpful if you can update the "help documents".

Hmmm...seems I digress,when I first use the conpot -t default,luckily it runs successfully.But "No Proxy template found.Service will remain unconfigured/stopped".I don't what's the meaning of it.Does it 
 impact on the communication between client and slave?Where is Proxy template is?

creolis, 2016-12-05:
Hej Kilento, thank you for your suggestions and your question.

Conpot supports a variety of protocols, like eg. SNMP, MODBUS, etc. Each of these protocols needs some sort of template in order to know what it should actually do. This is not about the protocol logic, it's about the data you want to simulate.

As you already noticed, we ship a "default" template. This template is acts as a basic showcase to give you a first impression on how conpot works, but of course you want to use your very own template. Using the default template in "production" is kind of senseless and will only waste your IP address for future use :-)

Providing some default values like .. idk .. the current system date / time, system load etc. is not a big deal for most protocols, so almost all of them are enabled by default in the template we ship.

The PROXY feature allows you to forward arbitrary traffic to e.g. a real system running next to conpot. You could for example relay traffic between conpot and a smart meter in order to see what's going on. It's like a flight recorder or .. let's call it a a simplified traffic analyzer.

Since you need to have something to forward your connections to, there is no simple way of enabling this feature in a meaningful way, so we did not provide any configuration for the proxy service in our default template.

Thus, you are notified on startup, that the proxy is not available unless you choose to configure it on your own :-)

hope that helps,

/)/)

Kilento, 2016-12-07:
Thanks a lot,I got it.It's very helpful for me to configure the conpot.

closed 2016-12-07 by Kilento