mushorg/conpot #317: Ubuntu 14 install

dirtdiver512, 2017-01-28:
Trying to get it to install on ubuntu 14. It seems to install but when I run it i get this error:

sudo conpot --template default
Traceback (most recent call last):
  File "/usr/local/bin/conpot", line 47, in <module>
    from conpot.protocols.bacnet.bacnet_server import BacnetServer
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 93, in __import__
    result = _import(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/conpot/protocols/bacnet/bacnet_server.py", line 26, in <module>
    from bacpypes.app import LocalDeviceObject
ImportError: cannot import name LocalDeviceObject

xandfury, 2017-01-28:
Have you tried from `bacpypes.app import LocalDeviceObject` in the python shell? Also what does `pip list` say?

xandfury, 2017-01-28:
Works fine in `Ubuntu 16.04.1 LTS`.

Vingaard, 2017-01-29:
just to check the conpot installation is good; 
have you tried running conpot with other templates, e.g ipmi or kamstrup_382

xandfury, 2017-02-01:
You may try docker installation. #318 
PR isn't merged yet though.

is161523, 2017-02-11:
If you have a bacpypes-Version greater than 0.13.8 try to uninstall bacpypes and install v0.13.8. That worked for me.

```
pip uninstall bacpypes
pip install -Iv https://github.com/JoelBender/bacpypes/archive/v0.13.8.tar.gz
```

closed 2017-02-13 by glaslos