mushorg/conpot #326: Testing conpot with the following tools/scripts

xandfury, 2017-06-09:
Testing is essential for conpot for it to emulate the "real thing"

Following scripts/tools might be helpful for the same

- https://github.com/enddo/smod
- https://github.com/drainware/nmap-scada

xandfury, 2017-12-18:
Also test with
https://github.com/0xICF/SCADAShutdownTool/blob/master/README.md

xandfury, 2017-12-18:
https://github.com/0xICF/ICScanner

Vingaard, 2017-12-18:
Hello @xandfury - please not that the two last suggestions-  (SCASAShutdown/ICScanner) - does not have the actual tool avaliable on github (and niether avaliable on the authors website) .  A shame, as both would have been interesting to test :-)

xandfury, 2017-12-19:
@Vingaard Code is in the beta branch!

https://github.com/0xICF/SCADAShutdownTool/tree/SCADAShutdownTool-v1.0-Beta?files=1

I'll find the link to ICScanner as well. Post it here as soon I find them

xandfury, 2017-12-26:
Results for some of scripts:

[1] -  From https://github.com/drainware/nmap-scada
***Siemens-Scalance-module*** : Execution fails due of undeclared snmpwalk function in the script.
```
Nmap done: 1 IP address (1 host up) scanned in 3.67 seconds
abhinav@abhinav-HP-ProBook-445-G1:~/projects/test/nmap-scada$ sudo nmap -sU --script=Siemens-Scalance-module localhost

Starting Nmap 7.01 ( https://nmap.org ) at 2017-12-26 14:19 IST
Nmap scan report for localhost (127.0.0.1)
Host is up (0.00010s latency).
Not shown: 992 closed ports
PORT      STATE         SERVICE
68/udp    open|filtered dhcpc
69/udp    open|filtered tftp
111/udp   open          rpcbind
161/udp   open          snmp
|_Siemens-Scalance-module: ERROR: Script execution failed (use -d to debug)
623/udp   open|filtered asf-rmcp
631/udp   open|filtered ipp
5353/udp  open|filtered zeroconf
47808/udp open|filtered bacnet
```

[2] - Smod - https://github.com/enddo/smod
***readCoilsException module*** : *Fails* - Output from smod
```
SMOD > use modbus/function/readCoilsException
SMOD modbus(readCoilsException) > show options
 Name     Current Setting  Required  Description                                 
 ----     ---------------  --------  -----------                                 
 Output   True             False     The stdout save in output directory         
 RHOSTS                    True      The target address range or CIDR identifier 
 RPORT    502              False     The port number for modbus protocol         
 Threads  1                False     The number of concurrent threads            
 UID                       True      Modbus Slave UID.                           
SMOD modbus(readCoilsException) > set RHOSTS 192.168.1.3
SMOD modbus(readCoilsException) > set UID 2
SMOD modbus(readCoilsException) > exploit
[+] Module Read Coils Exception Function Start
[+] Connecting to 192.168.1.3
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/abhinav/projects/conpot-test/smod/Application/modules/modbus/function/readCoilsException.py", line 63, in do
    ans = c.sr1(ModbusADU(transId=getTransId(),unitId=int(self.options['UID'][0]))/ModbusPDU01_Read_Coils_Exception(),timeout=timeout, verbose=0)
  File "/usr/local/lib/python2.7/dist-packages/scapy/supersocket.py", line 52, in sr1
    a,b = sendrecv.sndrcv(self, *args, **kargs)
  File "/usr/local/lib/python2.7/dist-packages/scapy/sendrecv.py", line 133, in sndrcv
    r = pks.recv(MTU)
  File "/usr/local/lib/python2.7/dist-packages/scapy/supersocket.py", line 125, in recv
    raise socket.error((100,"Underlying stream socket tore down"))
error: (100, 'Underlying stream socket tore down')
```
-----------------------------------------------------------------------------------------
Output from *Conpot*
```
2017-12-26 17:49:29,274 New modbus session from 127.0.0.1 (f728b030-664e-4794-b859-a37c322db864)
2017-12-26 17:49:29,276 New Modbus connection from 127.0.0.1:48054. (f728b030-664e-4794-b859-a37c322db864)
ERROR:conpot.protocols.modbus.slave:Exception caught: Modbus Error: Exception code = 1. (A proper response will be sent to the peer)
2017-12-26 17:49:29,280 Exception caught: Modbus Error: Exception code = 1. (A proper response will be sent to the peer)

```

xandfury, 2017-12-26:
Note that Conpot also issues ***ERRORS*** when tested with **Smod's DOS Scripts**
Sample Output when tested with Smod's ***modbus/dos/writeSingleCoils*** script 
```
ERROR:conpot.protocols.modbus.slave:Exception caught: Modbus Error: Exception code = 2. (A proper response will be sent to the peer)
2017-12-26 18:00:00,410 Exception caught: Modbus Error: Exception code = 2. (A proper response will be sent to the peer)
2017-12-26 18:00:00,411 Modbus traffic from 127.0.0.1: {'function_code': 5, 'slave_id': 2, 'request': '01fe00000006020537930000', 'response': '8502'} (f728b030-664e-4794-b859-a37c322db864)
2017-12-26 18:00:00,415 Modbus response sent to 127.0.0.1
2017-12-26 18:00:00,415 New Modbus connection from 127.0.0.1:49576. (f728b030-664e-4794-b859-a37c322db864)
ERROR:conpot.protocols.modbus.slave:Exception caught: Modbus Error: Exception code = 2. (A proper response will be sent to the peer)
2017-12-26 18:00:00,416 Exception caught: Modbus Error: Exception code = 2. (A proper response will be sent to the peer)
2017-12-26 18:00:00,420 Modbus traffic from 127.0.0.1: {'function_code': 5, 'slave_id': 2, 'request': '01ff000000060205d3750000', 'response': '8502'} (f728b030-664e-4794-b859-a37c322db864)
```

xandfury, 2017-12-26:
***SMOD modbus(readCoils)*** : StartAddress has been changed to 0x0001 as per the template.
```
SMOD modbus(getfunc) > use modbus/function/readCoils
SMOD modbus(readCoils) > set RHOSTS 127.0.0.1
SMOD modbus(readCoils) > show options
 Name       Current Setting  Required  Description                                 
 ----       ---------------  --------  -----------                                 
 Output     True             False     The stdout save in output directory         
 Quantity   0x0001           True      Registers Values.                           
 RHOSTS     127.0.0.1        True      The target address range or CIDR identifier 
 RPORT      502              False     The port number for modbus protocol         
 StartAddr  0x0001           True      Start Address.                              
 Threads    1                False     The number of concurrent threads            
 UID        1                True      Modbus Slave UID.                           
```
-------------------------
Output from ***smod***
```
SMOD modbus(readCoils) > exploit
[+] Module Read Coils Function Start
[+] Connecting to 127.0.0.1
[+] Response is :
###[ ModbusADU ]###
  transId   = 0x29f
  protoId   = 0x0
  len       = 0x4
  unitId    = 0x1
###[ Read Coils Answer ]###
     funcCode  = 0x1
     byteCount = 1L
     coilStatus= [0]
SMOD modbus(readCoils) > 
```
--------------------------
Output from ***Conpot***
```
2017-12-26 18:30:13,759 New Modbus connection from 127.0.0.1:50990. (87d51e98-6105-42cc-8709-aefb1198251b)
2017-12-26 18:30:13,770 Modbus traffic from 127.0.0.1: {'function_code': 1, 'slave_id': 1, 'request': '029f00000006010100010001', 'response': '010100'} (87d51e98-6105-42cc-8709-aefb1198251b)
2017-12-26 18:30:13,771 Modbus response sent to 127.0.0.1
ERROR:conpot.protocols.modbus.modbus_server:Exception occurred in ModbusServer.handle() at sock.recv(): timed out
2017-12-26 18:30:18,772 Exception occurred in ModbusServer.handle() at sock.recv(): timed out
2017-12-26 18:30:18,773 Modbus client disconnected. (87d51e98-6105-42cc-8709-aefb1198251b)
```

xandfury, 2017-12-27:
Unfortunately I couldn't get the `SCADAShutdown` tool to work. Also couldn't find code for `ICScanner`

xandfury, 2018-01-01:
This look interesting. Can check for Modbusm S7comm and ENIP 
https://github.com/dark-lbp/isf