mushorg/conpot #330: Investigate SNMP dependencies

glaslos, 2017-06-26:
pysmi and pysnmp should be pure python and we might be able to remove the libsmi2ldbl snmp-mibs-downloader dependencies.

windflyer, 2017-06-27:
`libsmi2ldbl` can be removed. We still need `snmp-mibs-downloader` in order to compile mib files.

glaslos, 2017-06-27:
Shouldn't https://github.com/etingof/pysmi be able to compile the MIB files?

windflyer, 2017-06-27:
You're right. I figured out how to completely get rid of `snmp-mibs-downloader`. A brief explanation:
When compiling our own MIB files we need to download dependencies from a remote repo (e.g., http://mibs.snmplabs.com/asn1/). We're now using `snmp-mibs-downloader` for doing so. 

We can let `pysmi` automatically download dependencies from a remote repo while compiling a MIB file. I'm gonna update the source code.

t3chn0m4g3, 2017-06-27:
This is really good news. Brings me one step closer to use Alpine for my docker image :)

creolis, 2017-06-27:
This sounds great and will also remove bloat! Go for it :)

windflyer, 2017-06-27:
See https://github.com/mushorg/conpot/pull/333

glaslos, 2017-06-28:
Fixed by eca2891

closed 2017-06-28 by glaslos