mushorg/conpot #331: Resberry PI

tzvetkov75, 2017-06-26:
Hi, 
I try to install conport on rasbery pi (banna pi) and it fails with some dependency. Can someone help.

``
vesselin@minipi:~/work/conpot$ conpot
Traceback (most recent call last):

  File "/usr/local/bin/conpot", line 47, in <module>
    from conpot.protocols.bacnet.bacnet_server import BacnetServer
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 93, in __import__
    result = _import(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/conpot/protocols/bacnet/bacnet_server.py", line 26, in <module>
    from bacpypes.app import LocalDeviceObject
ImportError: cannot import name LocalDeviceObject
vesselin@minipi:~/work/conpot$ 
``

I think I have all dependencies, like bacpypes.app 

``
vesselin@minipi:~/work/conpot$ sudo pip install conpot
[sudo] password for vesselin: 
Requirement already satisfied: conpot in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied: bottle in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: sphinx in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: pyghmi in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: requests in /usr/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: MySQL-python in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: crc16 in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: enum in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: stix in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: modbus-tk in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: beautifulsoup4 in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: hpfeeds in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: bacpypes in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: pysnmp==4.2.5 in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: gevent>=1.0 in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: jinja2 in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: stix-validator in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: libtaxii>=1.1.0 in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: lxml in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: cybox in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: xlrd in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: mixbox in /usr/local/lib/python2.7/dist-packages (from conpot)
Requirement already satisfied: docutils>=0.11 in /usr/local/lib/python2.7/dist-packages (from sphinx->conpot)
Requirement already satisfied: six>=1.5 in /usr/lib/python2.7/dist-packages (from sphinx->conpot)
Requirement already satisfied: sphinxcontrib-websupport in /usr/local/lib/python2.7/dist-packages (from sphinx->conpot)
Requirement already satisfied: typing; python_version < "3.5" in /usr/local/lib/python2.7/dist-packages (from sphinx->conpot)
Requirement already satisfied: alabaster<0.8,>=0.7 in /usr/local/lib/python2.7/dist-packages (from sphinx->conpot)
Requirement already satisfied: snowballstemmer>=1.1 in /usr/local/lib/python2.7/dist-packages (from sphinx->conpot)
Requirement already satisfied: babel!=2.0,>=1.3 in /usr/local/lib/python2.7/dist-packages (from sphinx->conpot)
Requirement already satisfied: imagesize in /usr/local/lib/python2.7/dist-packages (from sphinx->conpot)
Requirement already satisfied: setuptools in /usr/local/lib/python2.7/dist-packages (from sphinx->conpot)
Requirement already satisfied: Pygments>=2.0 in /usr/local/lib/python2.7/dist-packages (from sphinx->conpot)
Requirement already satisfied: pycrypto>=2.6 in /usr/local/lib/python2.7/dist-packages (from pyghmi->conpot)
Requirement already satisfied: python-dateutil in /usr/local/lib/python2.7/dist-packages (from stix->conpot)
Requirement already satisfied: pyserial>=3.1 in /usr/local/lib/python2.7/dist-packages (from modbus-tk->conpot)
Requirement already satisfied: pyasn1>=0.1.2 in /usr/local/lib/python2.7/dist-packages (from pysnmp==4.2.5->conpot)
Requirement already satisfied: greenlet>=0.4.10 in /usr/local/lib/python2.7/dist-packages (from gevent>=1.0->conpot)
Requirement already satisfied: MarkupSafe>=0.23 in /usr/local/lib/python2.7/dist-packages (from jinja2->conpot)
Requirement already satisfied: ordereddict in /usr/local/lib/python2.7/dist-packages (from stix-validator->conpot)
Requirement already satisfied: ordered-set in /usr/local/lib/python2.7/dist-packages (from mixbox->conpot)
Requirement already satisfied: weakrefmethod>=1.0.3; python_version < "3.4" in /usr/local/lib/python2.7/dist-packages (from mixbox->conpot)
Requirement already satisfied: pytz>=0a in /usr/local/lib/python2.7/dist-packages (from babel!=2.0,>=1.3->sphinx->conpot)

Linux minipi 4.9.7-sunxi #1 SMP Thu Feb 2 01:52:06 CET 2017 armv7l GNU/Linux

``

tzvetkov75, 2017-06-26:
I found the solution the version of bacpypes was not exactly matching

`sudo pip install bacpypes==0.13.8`

Can you handle this in setup or similar, since it can be very annoying. If not close the issue 
Thanks
Vesslin

glaslos, 2017-06-27:
One solution would be adding `--upgrade --force-reinstall` to the pip install command parameters.

tzvetkov75, 2017-06-29:
Many thanks :-)

xandfury, 2017-06-30:
Could have been a network error. Check [this](https://github.com/mushorg/conpot/blob/d97a68a054e4fe42ff90293188a5702ce8ab09a3/requirements.txt#L20). Requirement file specifies the exact version.

tzvetkov75, 2017-07-04:
yes, this is strange. I repeated this twice on 
`ARMBIAN 5.25 stable Debian GNU/Linux 8 (jessie) 4.9.7-sunxi` 
and have network connection. Assume it is my local environment problem.
Thanks

closed 2017-07-04 by tzvetkov75